Ultimaker/Cura #7411: Cura 4.5 crashing during start

Skifi, 2020-04-04:
**Application version**
4.5, 4.2.1, 4.0.0

**Platform**
AMD Ryzen 9 3900X
32GB RAM
NVIDIA GeForce RTX 2080 SUPER - latest drivers 445.75
Windows 10 Version 1909 build 18363.720 all updates installed.

**Printer**
No Printer, clean install and first run.

**Reproduction steps**
1. Installed Windows 10
2. Installed Latest Drivers for HW
3. After burn In HW test started installing new apps
4. Cura Installs without problems, but when i try to run it. It crashes without error.
5. Tryed run as Administrator, compatability to W7, W8.
6. Installed older version of Cura with manually removing AppData/Roaming/Cura folder for each install but same result.
**Screenshot(s)**

**Actual results**
App dont start

**Expected results**
Working Cura app.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
2020-04-04 12:36:24,420 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin ConsoleLogger
2020-04-04 12:36:24,432 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraDrive
2020-04-04 12:36:24,447 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: C:\Program Files\Ultimaker Cura 4.5\CuraEngine.exe
2020-04-04 12:36:24,450 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraEngineBackend
2020-04-04 12:36:24,463 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraProfileReader
2020-04-04 12:36:24,465 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [173]: Socket state changed to Listening
2020-04-04 12:36:24,474 - INFO - [MainThread] UM.Backend.Backend.startEngine [81]: Started engine process: C:\Program Files\Ultimaker Cura 4.5\CuraEngine.exe
2020-04-04 12:36:24,476 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [101]: [Backend] Calling engine with: ['C:\\Program Files\\Ultimaker Cura 4.5\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
2020-04-04 12:36:24,483 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin CuraProfileWriter
2020-04-04 12:36:24,490 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin OBJReader
2020-04-04 12:36:24,497 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin OBJWriter
2020-04-04 12:36:24,509 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin STLReader
2020-04-04 12:36:24,517 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin STLWriter
2020-04-04 12:36:24,525 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FileLogger
2020-04-04 12:36:24,533 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-04 12:36:24,535 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] Cura_SteamEngine version 4.5.0
2020-04-04 12:36:24,537 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] Copyright (C) 2019 Ultimaker
2020-04-04 12:36:24,540 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-04 12:36:24,542 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] This program is free software: you can redistribute it and/or modify
2020-04-04 12:36:24,544 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] it under the terms of the GNU Affero General Public License as published by
2020-04-04 12:36:24,546 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] the Free Software Foundation, either version 3 of the License, or
2020-04-04 12:36:24,548 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] (at your option) any later version.
2020-04-04 12:36:24,550 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-04 12:36:24,552 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdateChecker
2020-04-04 12:36:24,554 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] This program is distributed in the hope that it will be useful,
2020-04-04 12:36:24,556 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [177]: Socket state changed to Connected
2020-04-04 12:36:24,558 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2020-04-04 12:36:24,560 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [167]: Backend connected on port 49674
2020-04-04 12:36:24,563 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2020-04-04 12:36:24,566 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] GNU Affero General Public License for more details.
2020-04-04 12:36:24,568 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] 
2020-04-04 12:36:24,570 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] You should have received a copy of the GNU Affero General Public License
2020-04-04 12:36:24,572 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [101]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2020-04-04 12:36:24,579 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin FirmwareUpdater
2020-04-04 12:36:24,586 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzReader
2020-04-04 12:36:24,592 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeGzWriter
2020-04-04 12:36:24,599 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeProfileReader
2020-04-04 12:36:24,612 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeReader
2020-04-04 12:36:24,620 - INFO - [MainThread] UM.Logger.info [112]: Loaded plugin GCodeWriter
[... 390 lines cut ...]
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00004f80 (most recent call first):
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 370 in exec_
  File "X:\4.5-exe\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 823 in run
  File "X:\4.5-exe\build\inst\bin\cura_app.py", line 214 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load

**Additional information**
After runing app it starts loading but after while it shows blank window which imediatly dissapear and app crash without any error.
Windows is clean installed and updated. On my Intel laptop Cura works fine.

Ghostkeeper, 2020-04-07:
Hmm, I have pretty much exactly the same set-up as your computer. Same CPU. Only the 2070RTX graphics card instead of your nice 2080 but it's the same product line, and 64GB RAM instead of 32, but I'm not seeing any crashes on Windows.

Your log file also doesn't give a lot of information :( It's basically crashing when it's creating the main window. The traceback says that Qt is crashing during the main event loop. Nothing else. :confused:

Skifi, 2020-04-09:
@Ghostkeeper Thx for checking.

I think i just found my problem.
Just now i was backing up my files for new system reinstall. And got idea, I didnt try to disable my antivir sw. I am using Bitdefender. Its not showing any sign of problem but when i disabled it and tryed again to install Cura. Its working without problem.

So if anyone have similar problem. Just for installation try to disable your antivir sw.

closed 2020-04-09 by Skifi