Ultimaker/Cura #7412: [4.5.0] Slicer generates z-jumps / switches layer to build plate in mid print

mnemocron, 2020-04-04:
**Application version**
4.5.0

**Platform**
Windows 10 Pro 1903

**Printer**
Anet A8

**Reproduction steps**
1. use file `arm_part_2.stl` (see project file)
2. Select bottom surface to stick to build plate
3. Rotate part by 45° on the y-Axis 
4. Slice, using the default "Fine 0.1mm" Profile
5. Review in the "Preview" Tab

**Screenshots**
[image]

[image]

**Actual results**
The slicer shows that an intermediate layer is to be printed on the ground plane.
The print head would knok the wip-print off the bed.

**Expected results**
Continuous upwards layer transition from ground plane (z=0)

**Project file**
[CFFFP_arm_part_2.zip](https://github.com/Ultimaker/Cura/files/4431737/CFFFP_arm_part_2.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4431733/cura.log)

**Additional information**
It is difficult to reproduce. I wasn't able to by following the top steps.
I changed the profile to a different one (Keep customization), sliced, turned out fine.
Changed back to "Fine 0.1mm", sliced and now it suddenly didn't show the artifacts.

Nevertheless, I wanted to let someone know that this happened. I only noticed because I am using a color palette for the preview. If it was all yellow, someone might not see this.

Ghostkeeper, 2020-04-07:
I've seen this issue many times. It's not actually printing there though. It's only a visual glitch. You can verify this by saving the g-code to a file and loading it back in. The problem should be gone then. Or you can simply restart the slice (e.g. check/uncheck one setting or something).

The problem is somewhere in that the frond-end forgets the Z coordinate of one layer or the back-end forgets to send it, and it'll take a Z coordinate of 0 then. Only it happens VERY rarely. For me, working full-time as Cura developer, I see this maybe once a month. That makes it very hard to debug this issue.

fvrmr, 2021-01-11:
Reproduction steps of this issue: https://github.com/Ultimaker/Cura/files/5789052/MissingZLayers.zip

fieldOfView, 2021-04-11:
This is actually a duplicate of https://github.com/Ultimaker/Cura/issues/3815 which was never fixed

fvrmr, 2021-04-13:
We have a ticket on our backlog for this issue.
Developers see CURA-7960.

> This is actually a duplicate of #3815 which was never fixed

We will leave this ticket open since in this ticket is linked to our ticket on our backlog.

GregValiant, 2024-12-15:
This problem appears to have been resolved in 5.9.0 and I'll go ahead and close this.  Let me know if that is premature.
[image]

closed 2024-12-15 by GregValiant