Ultimaker/Cura #7415: Cura 4.4 & 4.5 cause the nozzle to lower before going to the print start point when Z-hop is enabled

MrD-RC, 2020-04-05:
**Application version**
4.4 & 4.5

**Platform**
Windows 10 - i7330K RTX2080

**Printer**
Prusa i3

**Reproduction steps**
1. Enable z-hop

Since Cura 4.4, some G-code is added somewhere that lowers the nozzle before going to the start point of the print job. Cura 4.3 and previous did not do this. The issue is that on some jobs, lowering the nozzle before getting to the start point will hit my nozzle in to the crocodile clips that hold the glass bed in place. It also means that the nozzle is lowered in to the extruded waste material from the cleaning process.

I know that it's not in the printer or extruder G-code in the machine settings. After adding [a post in the ultimaker forums](https://community.ultimaker.com/topic/31879-stop-cura-44-45-lowering-the-nozzle-before-printing/), someone checked out the project file and determined that it was the z-hop setting that is causing the problem. After turning off z-hop, I can confirm that this is where the problem lies.

Just for reference, here is my printer start g-code.

```
M117 Initialising...
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G92 E0 ;zero the extruded length
M117 Cleaning... ;Put cleaning message on screen
; - reposition nozzle
G1 Y10 Z5 F10000 ;raise nozzle 5mm and move along Y by 10mm
; - continue cleaning
G1 Z1 ;Move nozzle close to the bed
G1 E10 ;extrude 10mm of filament
G1 Z15 F12000 E7 ;Move up 15mm fast, while extruding 5mm
G92 E0 ;zero the extruded length again
G1 X0 Y0 F500 ;home the nozzle
M117 Printing...
```

I added the move between the reposition nozzle comments to avoid the nozzle lowering back down in to the waste filament. But, it shouldn't be needed. I've attached a project file where the error occurs.

[Visor_NHS.zip](https://github.com/Ultimaker/Cura/files/4433497/Visor_NHS.zip)

smartavionics, 2020-04-05:
Devs, I think the problem here is caused by this code in LayerPlan::addTravel()

```
    if (is_first_travel_of_layer)
    {
        bypass_combing = true; // first travel move is bogus; it is added after this and the previous layer have been planned in LayerPlanBuffer::addConnectingTravelMove
        first_travel_destination = p;
        first_travel_destination_is_inside = is_inside;
        if (layer_nr == 0 && retraction_enable && retraction_hop_enabled)
        {
            path->retract = true;
            path->perform_z_hop = true;
        }
        forceNewPathStart(); // force a new travel path after this first bogus move
    }
```

So setting perform_z_hop to true for that initial travel move on layer 0 is what is causing the nozzle to descend before moving to the print start position. I think that perhaps not doing a z-hop for the initial travel move could be better because there shouldn't be anything on the build plate to avoid so the travel move can start at the height that is specified in the start gcode.

Ghostkeeper, 2020-04-08:
There will be stuff there though for some printers, such as a prime line that people created in their start g-code, or even a clip or a sticker on the build plate. Problem is that we don't really know if this move is going up or down because the position after the start g-code is unknown. Cura currently assumes that it's [0,0,0] which is close enough for most start g-codes, but in your case it's [0,0,15].

MrD-RC, 2020-04-08:
To me that doesn’t make sense Ghostkeeper. If the end of the start gcode for someone’s printer has the nozzle low or high, it makes no difference. It doesn’t even really matter where the X or Y axis are either. 

If you turn off Z-hop, the first move that Cura does is move to the start of the print. Regardless of where the nozzle is in 3D space, it is a linear move of all axis at the same time. So as the X and Y move to the start, the nozzle slowly lowers to the start point. This is how it worked in 4.3 and below and how it works now in 4.4 and above with Z-hop disabled. This is how it should work always. This is absolutely fine. 

The problem only occurs when Z-hop is enabled. It adds a movement where there is no need to. Smartavionics, I believe has the technicalities issue, I’ve never looked at the Cura code. There’s no need to add a z-hop at the start of the print as there has not been any filament extruded that needs to be hopped over. Stopping that first z-hop before the job really even starts would fix the bug.

smartavionics, 2020-04-08:
> There’s no need to add a z-hop at the start of the print as there has not been any filament extruded that needs to be hopped over.

Except that Cura doesn't "know" what the initial height is so if the start gcode sends the nozzle to z = 0 and it then doesn't use z-hop, the travel to the first print location will occur at z=0 which could cause a scratch on the print bed, tape, etc.

If the start gcode sets z to something > 0, then moving to z-hop height is not required.

smartavionics, 2020-04-08:
Personally, I would zap the z-hop on that first move in the code snippet above and tell people that they should make z > 0 in the start gcode.

MrD-RC, 2020-04-17:
I’d just like to add that what is the chance of the nozzle being lower than 0? If there is no chance of this, then why worry about the Z to 0 at all in the first move. 

Also, it doesn’t do this behaviour unless z hop is active. Therefore the z hop is inconsistent with how the first move should work. It would be better if the first move by the printer was consistent, therefore something that can be planned for.

Ghostkeeper, 2020-04-22:
I don't think the problem is that the nozzle has a lower coordinate than 0, but the problem is that the nozzle has a lower coordinate than the height of the bed, which may not be aligned to 0. If the nozzle moves below the print head of the bed, you'll get a nasty scratch. So really it's a problem if the nozzle is lower than the initial layer height, which is sort of guaranteed to be at a safe distance (in fact, for Ultimaker printers we set the initial layer height as high as the nozzle can take in order to deal better with badly levelled beds).

MrD-RC, 2020-04-23:
That’s fair enough. But if the nozzle is lower than the first layer height or the glass, this would be an odd setup, maybe to chop filament? In which case, they would have something in the start gcode to raise the nozzle. If not, they would experience problems on prints that don’t have z-hop enabled.

Ghostkeeper, 2020-04-28:
Some printers dip to 0 in order to probe the bed height. Some dip to 0 in order to attach their prime line to the build plate. In either case they don't experience problems since the travel move towards the start of the print is safe.

BryceDFisher, 2020-08-10:
I'm also having these same issues with Cura 4.6 on my Ultimaker 2 with z-hop enabled.  It's causing the nozzle to lower into the waste plastic at the start of a print, which causes the waste plastic to stick to the nozzle, which then causes me to abort the print.

Ghostkeeper, 2020-08-12:
Yes, this was fixed in Cura 4.7 for the Ultimaker 2 family: https://github.com/Ultimaker/Cura/commit/be814535e979c28734460a697da879dbe2b41a07

The firmware places a prime blob in a specific location, unbeknownst to Cura. Cura needs to move away from that location because it is essentially restricted terrain.

GregValiant, 2026-05-30:
Is this still a problem in 5.12.1 or can this be closed?