Ultimaker/Cura #7416: Dual Extrusion - Ooze Shield - number of perimeters

SpectroMaledikto, 2020-04-05:
When I enable Ooze Shield, Cura build only one perimeter.
Sometimes the Ooze Shield detach from bed...and the print is aborted.

Can you insert function '' Ooze Shiel number of perimeters ''

This option makes it more functional this function because reduce
the probability of detaching.

Grazie.

Ghostkeeper, 2020-04-08:
The ooze shield gets the same bed adhesion as support and any other feature though. If you're having trouble with bed adhesion, why are you not using a brim, or a raft?

Here's a screenshot that displays the brim attached to an ooze shield:
[image]

SpectroMaledikto, 2020-04-10:
why not 2 or 3 perimeters ?
with a 0.3mm hotend the perimeter is very fragile (breakable)

Ghostkeeper, 2020-04-14:
Oh, it's about fragility, not adhesion.

I guess we've never really considered that the ooze shield needs to be strong in any metric. It just needs to capture the ooze and that doesn't weigh much. I must admit that I've never tried this with a 0.25mm nozzle though. I use a prime tower instead, then.