Ultimaker/Cura #7419: [4.5] Weird infill path with different Layer Heigh

edseUNAD, 2020-04-05:
**Application version**
4.5

**Platform**
Windows 10
Nvidia GTX1050Ti

**Printer**
Custom Printer, converted to DirectDrive Ender3 

**Reproduction steps**
1. Slice the model with 0.2 LH
2. Preview lfirst ayer, playing it from beginning shows weird infill path

**Screenshot(s)**

Issue when slicing at 0.2
[image]

middel layer

[image]

Upper layer
[image]

**Actual results**
First layers seems ok, but playing it from beginning shows weird infill path, seems like it is reflecting upper parts of the model, changed LH to 0.1 and the issue is not observed, path is straight lines

**Expected results**
First layer infill should be straight lines without any kind of geometry

**Project file**
bullet_bang.3mf
[bullet_bang.zip](https://github.com/Ultimaker/Cura/files/4434525/bullet_bang.zip)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-08:
This is indeed easy to reproduce. I couldn't find anything wrong with the model. Disabling gap filling doesn't make a difference. It's just weird. It seems to think that it needs to print skin there, except that it doesn't produce an outline around this part of skin either.

I think we might have an edge case on our hand here. Either way, we've put this in our backlog to investigate for a bugfix.

smartavionics, 2020-04-08:
With top surface skin layers set to 1...

[image]

konskarm, 2020-04-08:
Developers see CURA-7348

MrEd88, 2020-04-08:
> With top surface skin layers set to 1...
> 
> [image]

Using 1 skin layer the issue is reproduced in the second layer

[image]

Using 0 skin layers seems to completelly solve the issue. Is it still a bug? Or is there any rule of how many skin layers to use?

Ghostkeeper, 2020-04-09:
Oh, that's a good find, Smartavionics. Indeed the top surface gets separated from the rest of the skin. That's why it's printing those parts separately.

The "top surface" is a structure that has a different set of settings so that you can set the line width, speed, acceleration, flow, pattern and line directions differently for the very highest layer. This way you could for instance print the last layer more slowly than the rest of your skin, and achieve a smoother surface without needing to print ALL of the skin more slowly.

Only in your profile you have the speed, flow, line width, etc. set exactly the same as the top/bottom speed, flow, line width, etc. So there is no actual difference in how it prints or the print quality, but it will still separate out the different segments to print them with specialised settings. Just by coincidence those specialised settings are exactly the same as the normal settings. The shape of the bullet comes back there in those first few layers since there is a groove around the bullet shape that is just 3 layers above the bottom of the object.

I'd say this is expected behaviour then. Though in theory it could check if all of the settings are exactly the same as the top/bottom settings and then not apply the top/bottom surface layers at all. But that's a bit clunky I'd say and dangerous to future changes.

closed 2020-04-14 by Ghostkeeper