Ultimaker/Cura #7420: Material cost not seperate

TheNetWhiz, 2020-04-05:
**Application version**
(The version of the application this issue occurs with.)

4.5.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Windows 10 Pro

**Printer**
(Which printer was selected in Cura?)

Anet A8

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

Select a filament material and assign a price for the roll. Switch to another the the price is carried over. Change it to something different. Switch back to the first one and the price is changed to the new price.

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

Same price for all materials.

**Expected results**
(What should happen after the above steps have been followed.)

Each material should have its own price saved with it for cost calculations.

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-08:
This is what we used to have an "Unlink" button for but I don't see it any more. Where has that gone? Does anyone know?

Ghostkeeper, 2020-04-10:
Ah, no the "Unlink" button is still there. I wasn't looking properly. It's halfway down the scrollview just underneath the prices and weights:

[image]

Push that button and you should be able to configure the price and weight and other metadata of your duplicates separately.

Why it works like this is a bit technical. Basically we intended these duplicated material profiles to be used as profiles: A set of settings that you can switch easily between. But they are profiles for the same material. We keep track of which manufactured material it is. If you're using an Ultimaker or another printer that features automatically syncing the materials with what's on the printer, then Cura will select one of the material profiles that matches with the spool that's on the printer. Internally they match if the printer has the same [GUID](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/machine_settings/material_guid.md) as the profile. When you duplicate the material, the GUID is not changed so Cura could now select your duplicated profile as well when it syncs with your printer.

When you click the "unlink" button, Cura will change your GUID and so now it thinks it's a completely different spool. The different spool can have a different price and weight. And Cura will no longer allow that profile to be selected when syncing with the printer's configuration.

closed 2020-04-10 by Ghostkeeper

guate6, 2020-12-29:
Using Cura 4.8.0 and the "unlink" option is NOT there. All my different materials are linked and I can't accurately get project cost to display. Reached out to ultimaker support and was told to come here to comment. 
[image]

Ghostkeeper, 2021-01-06:
The button is not shown if the list of linked materials is empty, i.e. the material doesn't share properties with any other material.

I'm unable to get the button to disappear for a duplicated material, except by clicking the "Unlink Material" button. I can't reproduce your issue as far as I can see.

Perhaps one thing we could do is to load up your entire configuration. It's a bit of a hassle maybe, but could you go into Help -> Show Configuration Folder and then zip up that "4.8" folder and post it here? That might allow us to see if somewhere along the way something got out of sync in your material files or something.

guate6, 2021-01-07:
Each material I've input into the material profiles starts by me
clicking on "generic PLA" and duplicating it.
Then I go and customize the fields I need. I don't see anything about
"linked materials" anywhere. Please advise. Thanks.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

Ghostkeeper, 2021-01-18:
I'm still unable to reproduce that problem, and have no new information to try anything else. This is what I see when I click on the "Duplicate" button while Generic PLA is selected:

[image]

So I get a second Generic PLA. And in the bottom right it says "This material is linked to PLA and shares some of its properties". One of the properties it shares is cost. The intent of this is that you can have different profiles (settings) for the same spool of filament. So the settings can be different, but the metadata like spool weight and cost is not. And it shows an "Unlink" button there.

I believe you on your word that you're experiencing this issue, but I can't reproduce it here.

guate6, 2021-01-18:
I went back to look and here's what I found:
I click on the new material already in there - I click away, but have to
click back and once that's done the "unlink material" appears.
Thanks.

Raúl E. Fernández
Office (call/text): 352-573-6114
Mobile (call/text): 352-234-6161

Ghostkeeper, 2021-01-26:
Oh, that's strange. That shouldn't be necessary. After duplicating it, the button appears immediately for me in any case.