Ultimaker/Cura #7421: The extruder icon on the Preview tab looks like a notification icon

brettp, 2020-04-05:
**Is your feature request related to a problem? Please describe.**

Yes. For users with only 1 extruder, it is not obvious what the first extruder icon is supposed to be. It is very close to a common design element for pending notifications or information about the current settings.

Also, this area is used to display statuses for some icons (the green check for the selected printer), so it's confusing to see what looks like a notification for "1 invalid setting," even within the context of the material and nozzle information.

**Describe the solution you'd like**
This could be remedied by adding hover text to the dropdown element that says "Extruder 1," Extruder 2," etc. It could also be remedied by not displaying the number if only 1 extruder is used.

**Describe alternatives you've considered**
Googling for what this icon is and why it's always displaying a 1. (No helpful results.)

**Affected users and/or printers**
Users new to Cura who only have a single extruder, and most people, since we're all used to the notification badge UI element.

**Additional context**
Cura icon: 
[image]

Example unread notification / comment icon: 
[image]

Yian-Lai, 2020-04-06:
Hi, thanks for the feedback! We are working on improving the icons in Cura, so will take this into account together with other redesign.

brettp, 2020-04-06:
Thanks, @Yian-Lai. Will that redesign also take into account hover text? Also, is there an ETA or ticket to track that task?

Ghostkeeper, 2020-04-06:
I think so far it only includes the icons, not any texts. I expect the ETA somewhere around the start of the summer, but no promises. Ultimaker always has redesigns in the summer.

brettp, 2020-04-06:
Thanks. Is there a task / ticket to track this so folks can contribute if
able?

nallath, 2020-04-07:
Well, we have an internal tracking ticket for it. But maybe @Yian-Lai can already share some of the designs?

Ghostkeeper, 2020-04-08:
Having plans to fix this doesn't mean that it's fixed yet, so I'll reopen this until we have that done.