Ultimaker/Cura #7423: MAC OS X - pre-installed Material does not show up (generic PLA, ABS, etc.)

Timisageek, 2020-04-05:
**Application version**
4.5.0
**Platform**
MacOS X 10.14.6 (Mojave)

**Printer**
New Custom Printer (you guys don't support Flyingbear printers :'( )

**Reproduction steps**
1. Tried to select material : generic PLA by clicking on the menu
2. The drop down menu stays empty - nothing shows up
3. Went to the preferenceS/settings. It says all generic material is already installed, but nothing shows up :(

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-06:
I'm not able to reproduce these. I added a completely fresh Custom FFF Printer without any modifications. I went to the material selection drop-down and there I can see the Generic materials, Structur3D and Ultimaker brand materials.

Did you adjust the supported filament diameter in the printer settings dialogue? Are you sure you didn't make a typo there, like that it supports 175mm materials rather than 1.75mm?

gabork, 2020-04-12:
@Ghostkeeper Thanks for the 175mm tip, it was spot on for me! I had a very similar issue where the filament list was empty, and the filament setting page was not reacting to any buttons. Only, I was on Ubuntu 18.04, but otherwise the description matches.

The issue was that when I set up a custom printer, the filament diameter field was ignoring the dot/period key, without me noticing. It was even ignoring the key when I went back to double check. It might be a regional setting issue because some apps enforce the decimal comma instead of the decimal point (my machine's regional format mandates the comma). I ended up copying a decimal point from another field.

It would be a nice feature to have a small warning text on the filament page when the list is empty. Just by looking at it, it isn't obvious that it has a filtering logic at all, and if it's empty because there are no filament profiles in the binary package.

I can't speak for @Timisageek, and I'm using an OS different to his, but the filament diameter setting fixed the issue for me. Thanks again!

Ghostkeeper, 2020-04-21:
Hopefully we can replace it with our own widget soon. It's currently a system-styled widget there, which is why it pays particular attention to your system settings like the radix to use for your locale. Normally Cura always uses periods as radix and if you try to use a comma it changes it to a period.

closed 2020-06-19 by no-response[bot]