Ultimaker/Cura #7425: [4.3] Change heating sequence for Creality CR-X

c-strecker, 2020-04-06:
**Application version**
Cura 4.3

**Platform**
Windows

**Printer**
Creality CR-X

**Reproduction steps**
1. Creating gcode file to send to printer. Cura 
creality_cr-x.def.json file appends start gcode to have the hotend heat first. Custom start code has to be put in to switch heating to bed first.

**Actual results**
Hotend warms up before bed.

**Expected results**
Starting gcode generated by Cura should heat up bed first.

Ghostkeeper, 2020-04-08:
We don't really have a way to test these profile changes so we have to rely on our community to make them. Is there anyone with a CR-X that can verify that this is indeed a bug?

c-strecker, 2020-04-09:
Maybe this isn't a bug per se, and just a preference. I thought most printers heated up the bed first.

Ghostkeeper, 2020-04-14:
I'd also say it's better to heat up the bed first, then the nozzle, since the material will be oozing and degrading while the nozzle is hot. However some printers may not have enough power in their PSU to keep the bed hot while the nozzle is heating. The heaters for the bed are often much more power-hungry than the heaters for the nozzle.

c-strecker, 2020-04-15:
I have a CR-X. I opened it up the other day to check on the board due to temperature variations and it has a meanwell 500w power supply. Plenty of power for both at once even.  I am happy to wait to see if anyone else has input before changes are made. It's a simple custom gcode change, but I just have to remember to do it before prints.

closed 2020-06-19 by no-response[bot]