Ultimaker/Cura #7426: [4.5.0] Cannot sync with marketplace

automayt, 2020-04-06:
**Application version**
4.5.0

**Platform**
Windows 10

**Printer**
Ultimaker S5

**Reproduction steps**
1. Clicked marketplace and browsed to PETG. This required login. Added to Cura.
2. Restarted Cura, a popup mentioned needing to sync and restart Cura. 
3. Restarting Cura results in the popup occurring everytime and the PETG profile is not added.

**Screenshot(s)**
[image]
[image]
[image]

**Actual results**
Restarts just result in requiring another sync. Only going to marketplace and unsubscribing to the material will result in removing the "sync" popup.

**Expected results**
The material profile should be added without issue.

**Log file**
`2020-04-06 01:36:29,668 - DEBUG - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._handlePackageDiscrepancies [90]: Discrepancy found between Cloud subscribed packages and Cura installed packages
2020-04-06 01:36:33,768 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [d519dc7127eb4e4bb8a0e060ce76d7f1] finished.
2020-04-06 01:36:33,772 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:34,067 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [4370b26c3352428794a37209464091fc] finished.
2020-04-06 01:36:34,069 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:38,549 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-06 01:36:41,157 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Ultimaker Cura 4.5/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
2020-04-06 01:36:42,517 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [927bdf4a51bf464886a6d3e1f06174d1] started
2020-04-06 01:36:42,520 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:43,034 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [927bdf4a51bf464886a6d3e1f06174d1] finished.
2020-04-06 01:36:43,041 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:43,063 - DEBUG - [MainThread] UM.PackageManager.getPackageInfo [573]: Found potential package.json file '/package.json'
2020-04-06 01:36:43,066 - INFO - [MainThread] UM.PackageManager.installPackage [426]: Package [GenericPETG] version [1.0.1] is scheduled to be installed.
2020-04-06 01:36:43,073 - INFO - [MainThread] UM.PackageManager._saveManagementData [248]: Package management file C:\Users\jason\AppData\Roaming\cura\4.5\packages.json was saved
2020-04-06 01:36:43,077 - DEBUG - [MainThread] UM.Logger.debug [107]: Subscribing to GenericPETG
2020-04-06 01:36:43,092 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._processRequest [308]: Request [6997aebc108047fba3024a2c38694452] started
2020-04-06 01:36:43,094 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:43,347 - INFO - [MainThread] UM.TaskManagement.HttpRequestManager._onRequestFinished [369]: Request [6997aebc108047fba3024a2c38694452] finished.
2020-04-06 01:36:43,352 - DEBUG - [MainThread] UM.TaskManagement.HttpRequestManager._processNextRequestsInQueue [279]: No more requests to process, stop
2020-04-06 01:36:44,360 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [462]: Shutting down cura
2020-04-06 01:36:44,398 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [292]: Attempting to kill the engine process
2020-04-06 01:36:44,399 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [298]: Killing engine process
2020-04-06 01:36:44,400 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Engine process is killed. Received return code 1`

bberrevoets, 2020-04-08:
Hello,
Almost the same here:
Windows 10
Cura 4.5.0
Generic PETG
Ultimaker 3

But removing it from the marketplace in the webbrowser, does not get rid of the loop it is in.
It keeps asking to sync and restart the whole time to install the material.

Ghostkeeper, 2020-04-08:
I tried to reproduce this issue. My account has several plug-ins attached and one material. It asked to sync those when I logged in. After I restarted it had installed these plug-ins. And it didn't ask to sync again.

The log says that it scheduled the package to be installed upon next reboot. So far it looks like it's going well. But apparently something goes awry when it actually needs to install the package.

Ghostkeeper, 2020-04-08:
Oh, I see it now! It happens with materials that are already distributed by default with the release of Cura. Like the Generic materials or the Ultimaker materials.

Ghostkeeper, 2020-04-09:
Apparently our testers had already found this as well. We're looking into a workaround here such as preventing you from adding these to your account, or pre-adding them to the account and then ignoring them in the syncing.

PapaBalein, 2020-04-28:
Yup, I just got this as well. There doesnt seem to be any way of using the Tough PLA material (Ultimaker or Generic) as Cura seems to think it's installed, but it's not a viewable option when choosing the material. When I log into my account online, it seems to suggest it's not installed either, so when i choose to install it, I observe the same loop the OP mentions.

Ghostkeeper, 2020-04-28:
> but it's not a viewable option when choosing the material.

It should be an option, at least if your printer uses 2.85mm filaments.

This issue is still in our backlog. We might also see a fix coming from the Ultimaker web services team.

PapaBalein, 2020-04-29:
> the Ultimaker web services team, rather than in Cura.

It really is not. Would  you like me to provide screenshots?

Ghostkeeper, 2020-04-29:
No, I believe you. We could be running into a bug here, or maybe there's something weird in your current printer. A project file would be more useful than a screenshot. Though maybe a screenshot would help as well if I'm misunderstanding which list of profiles is empty exactly. I'm thinking it's just the materials list in the drop down in the drop down in the middle of the top bar.

PapaBalein, 2020-05-05:
> it's the materials list in the drop down in the drop down in the middle of

Hey, sorry for the wait.

So, seeing that I'm currently trying to print with Tough PLA, I'm having
issues with adhesion. I'm suspecting that the print temperature is a bit
too low...which would make sense because the print temperatures are about
10 degrees cooler with the PLA preset (which I'm using in lieu of the Tough
PLA setting). Any idea how I could get around this? I'm a bit new to
Cura....

[image: image.png]

Cheers,
Brent

PapaBalein, 2020-05-05:
OOOOOHHHHHHHHH...I think I know! Tough PLA is "experimental" only with the
Ultimaker 2+. Is it possible that it's hiding it from my view because of
that?

Cheers,
Brent

Ghostkeeper, 2020-05-07:
Tough PLA is indeed explicitly hidden for the Ultimaker 2+. See the following line in the Ultimaker 2+ definition:

https://github.com/Ultimaker/Cura/blob/23cdf871c1f6b1bb49311d295370314f2669fc14/resources/definitions/ultimaker2_plus.def.json#L16

This is because the Ultimaker 2+ firmware doesn't have a profile for Tough PLA, which needs different printing temperatures. Cura cannot set the printing temperatures for the Ultimaker 2+ due to the g-code flavour.

You have a different issue than the original reporter here then, because that was reported for the UMS5.

Ghostkeeper, 2020-05-11:
Yeah you're basically getting the same problem then, because the sync with the marketplace has nothing to do with which printer you're using. Installing it while you are using an UM2+ vs. installing it while you are using an UMS5 has the same issue.

I think the Ultimaker site is wrong on that one. There are profiles for TPLA that I added myself to the UM2+ firmware, but that version of the firmware never got released. If you were to download a firmware update now it will not have it.

PapaBalein, 2020-05-11:
Interesting.

semiconducted, 2020-05-13:
> Apparently our testers had already found this as well. We're looking into a workaround here such as preventing you from adding these to your account, or pre-adding them to the account and then ignoring them in the syncing.

Same on **macOS** 10.15.4
across Ultimaker Cura 4.5.0, 4.6.0, & 4.6.1
- tried complete uninstall, restarted computer, reinstall
- tried adding "Full Disk Access" in macOS security preferences
- explicitly added "allow incoming connections" to macOS firewall rules to both 4.5.0 and 4.6.0 as per [instructions from tech support](https://community.ultimaker.com/topic/31670-cura-will-not-sync-with-ultimaker-account/?do=findComment&comment=263346)
- happens on 2 different wifi networks and also via Verizon mobile tether, so doubtful that has anything to do with it.
- Issue has apparently been cross referenced with "Internal ref CURA-7425"

Hope this helps!

Log:
[cura.log](https://github.com/Ultimaker/Cura/files/4623590/cura.log)

Ghostkeeper, 2020-05-15:
This is now fixed in Cura 4.7 as per this change: https://github.com/Ultimaker/Cura/pull/7707

Thanks for the report! That really helps to iron out the kinks.

closed 2020-05-15 by Ghostkeeper

PapaBalein, 2020-05-15:
Awesome :)

Thanks for the amazing back and forth!

Cheers,
Brent