Ultimaker/Cura #7428: Support Blocker renders project unslicable

ScruffR, 2020-04-06:
**Application version**
4.5.0

**Platform**
MS Suface Pro 2017 (Intel Graphics)
Windows 10 Pro 64bit (Build 19555.1001) 

**Printer**
Creality CR-10 (dual extrusion mod with E3D Chimera)

**Reproduction steps**
1. Slicing the model without support blockers works fine
2. Adding a single support blocker causes the slice to fail (message: "Unable to slice" - no reason given)

**Screenshot(s)**
Original without blockers (from above)
[image]
(from below)
[image]
To remove the three tiny support structures added a single support blocker
[image]

**Actual results**
As mentioned above, Cura 4.5.0 is unable to slice

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
[MountWithFanShroud.zip](https://github.com/Ultimaker/Cura/files/4437453/MountWithFanShroud.zip)

**Log file**
```
2020-04-06 11:36:26,833 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [225]: Starting to slice...
2020-04-06 11:36:26,837 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [243]: Going to slice build plate [0]!
2020-04-06 11:36:29,277 - ERROR - [Thread-1] UM.Logger.logException [98]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x0000014A4D621208> caused an exception
2020-04-06 11:36:29,281 - ERROR - [Thread-1] UM.Logger.logException [102]: Traceback (most recent call last):
2020-04-06 11:36:29,284 - ERROR - [Thread-1] UM.Logger.logException [102]:   File "X:\4.5-msi\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 121, in run
2020-04-06 11:36:29,288 - ERROR - [Thread-1] UM.Logger.logException [102]:   File "C:\Program Files\Cura\plugins\CuraEngineBackend\StartSliceJob.py", line 260, in run
2020-04-06 11:36:29,292 - ERROR - [Thread-1] UM.Logger.logException [102]:     self.setMessage(", ".join(associated_disabled_extruders))
2020-04-06 11:36:29,296 - ERROR - [Thread-1] UM.Logger.logException [102]: TypeError: deprecated_decorator() takes 1 positional argument but 2 were given
2020-04-06 11:36:48,993 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [129]: Writing to [D:/MountWithFanShroud.3mf]...
2020-04-06 11:36:48,995 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [109]: 'info/send_slice_info' is turned off.
2020-04-06 11:36:49,093 - DEBUG - [MainThread] LinearAdvanceSettingPlugin.LinearAdvanceSettingPlugin._filterGcode [109]: No used extruders specify a linear advance factor
2020-04-06 11:36:49,094 - DEBUG - [MainThread] LocalFileOutputDevice.LocalFileOutputDevice.requestWrite [150]: Writing to Local File D:/MountWithFanShroud.3mf in binary mode
2020-04-06 11:36:52,898 - DEBUG - [Thread-2] UM.FileHandler.WriteFileJob.run [76]: Writing file took 3.7642054557800293 seconds
2020-04-06 11:36:53,533 - DEBUG - [MainThread] cura.Machines.Models.QualityProfilesDropDownMenuModel._update [63]: Updating QualityProfilesDropDownMenuModel.
2020-04-06 11:36:53,553 - DEBUG - [MainThread] cura.Machines.Models.CustomQualityProfilesDropDownMenuModel._update [34]: Updating CustomQualityProfilesDropDownMenuModel.
2020-04-06 11:37:03,014 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-06 11:37:23,009 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [398]: file:///C:/Program Files/Cura/resources/qml/RoundedRectangle.qml:68: TypeError: Cannot read property of null
```

**Additional information**
It would be nice to have a menu option (e.g. under `Help` where you can directly access the most recent log file)

nallath, 2020-04-06:
This should be fixed for the 4.6 beta. Could you try if it works for you now?

ScruffR, 2020-04-06:
I can give it a try.

Ghostkeeper, 2020-04-08:
Yeah, I was able to reproduce that bug pretty easily with your new information.

BagelOrb, 2020-07-13:
[image]
Same issue here.

Reproduce steps:
- disable left extruder
- load object and set to second extruder
- add a support blocker
- hit slice

result: [unable to slice]

Ghostkeeper, 2020-07-15:
Thanks for the ping. I'd brought this up with the team back in April and our conclusion is that we would add this to the top of our backlog, because we weren't able to fit it in that sprint but would hopefully be able to fix it the sprint after. I see now that that never got done. Slipped through the cracks as it were. And nobody reacted here either.

BagelOrb, 2020-07-15:
To be honest, it's not a big problem. It's annoying, but there's several work-arounds. It's also pretty clear as a user what is causing the bug.

mahtDFR, 2020-07-15:
This is now definitely on the dev backlog as CURA-7594