Ultimaker/Cura #7429: Cura 4.5.0 will not save Gcode when z offset is set due to float error

StanislavJochman, 2020-04-06:
**Application version**
4.5.0
**Platform**
Manjaro 19.0.2 Kyria
Kernel: x86_64 Linux 4.19.113-1-MANJARO
**Printer**
Does not matter

**Reproduction steps**
1. Installed z offset
2. Set z offset to 0.1mm
3. Sliced Gcode - everything looks fine but can not save gcode
4. Error in terminal see screenshot

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]
** Logs **
2020-04-06 13:45:35,281 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [144]:     adjusted_z = round(float(result.group(2)) + z_offset_value, 5)
2020-04-06 13:45:35,281 - ERROR - [MainThread] UM.Qt.Bindings.OutputDeviceManagerProxy._writeToDevice [144]: ValueError: could not convert string to float: ''
2020-04-06 13:45:39,218 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles

**Additional information**
If z offset is set to 0 it works fine but when it is set to 0.1 and I use 0.8 mm nozzle it throws this error

nallath, 2020-04-06:
Since this is an issue caused by the z_offset plugin, you might want to report it there instead. I don't think there is much we can do from this side.

fieldOfView, 2020-04-06:
Please open an issue over at https://github.com/fieldOfView/Cura-ZOffsetPlugin/issues and I'll get cracking.

StanislavJochman, 2020-04-06:
It is open there so I think I can close it here right?

closed 2020-04-06 by StanislavJochman