Ultimaker/Cura #7430: error G-code, stop the machine in midle

Dragon3DPrinter, 2020-04-06:
[image]
[image]

[rueda_cajon_cocina.zip](https://github.com/Ultimaker/Cura/files/4438429/rueda_cajon_cocina.zip)

mahtDFR, 2020-04-06:
Hello, can you please fill out the issue template that you deleted so we can understand the issue and try to help?

nallath, 2020-04-06:
Also, I don't see what is wrong with that g-code. It's a travel move and your machine should be capable of handling those.

Dragon3DPrinter, 2020-04-06:
> Hola, ¿puede completar la plantilla de problema que eliminó para que podamos entender el problema e intentar ayudarlo?

Printing starts and suddenly stops at the first layer without turning off motors or changing temperatures, so I assume the error is from the slicer.

Sorry I deleted the template

Dragon3DPrinter, 2020-04-06:
> Además, no veo lo que está mal con ese código g. Es un movimiento de viaje y su máquina debería ser capaz de manejarlos.

I think I cannot guarantee with certainty that the printers always work with code G1 work and not G0 translation.

Ghostkeeper, 2020-04-08:
If your printer doesn't support the `G0` command, I suggest switching the g-code flavour to "Bits from Bytes":

[image]

closed 2020-04-08 by Ghostkeeper