Ultimaker/Cura #7431: Time estimate for gyroid infill pattern significantly under-estimates the time

CTotten-Sonalysts, 2020-04-06:
I am using Cura 4.5 (Windows 10) with an Ultimaker S5 and 3E.  I have noticed over time that some prints take significantly longer than the Cura estimate.  Until recently, I wasn't sure why.  But I am not convinced that it is related to the gyroid infill pattern, which I have been using extensively since it was added to Cura.
The time issue was significantly worse on large prints with a lot of infill.  For instance, I printed a large mounting fixture a couple of weeks ago (~ 7in x 7in x 3in).  The original Cura estimate was 28 hours.  I came back 30.5 hours later to start another print job, but the S5 display showed 4.5 hours remaining.  This is what got me thinking about this again.  I had a large object that I needed to make 7 copies of.  So I printed the first using my standard setup, with gyroid infill at 25%.  Cura estimate was 65.63 hours.  It ended up taking 73.73 hours.  On the second print, the only setting I changed was to Tri-Hex infill at 25%.  Cura estimate for this print was 64.75 hours.  As of this moment, there is about 3.5 hours left on the print, but total time should come in at about 64.5 hours.
This print job is using UM CPE at 0.2mm layer height, and Breakaway infill (triangles at 15%).  The mounting fixture was UM Black Nylon at 0.2mm layer height.

Here is a link to the Ultimaker Community thread on the subject:
https://community.ultimaker.com/topic/31811-cura-time-estimation-using-gyroid-infill/
I posted the completion time graphs that I created during the two comparison print jobs, basically showing the estimated completion time (in hours) vs. the time I made the entry.

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

mahtDFR, 2020-04-06:
Would be helpful if you can save a project file (File -> Save...), zip up the .3mf file and attach to this issue so we can try to reproduce it.

CTotten-Sonalysts, 2020-04-06:
Attached.  Be advised it is a large print (~900g of material).  This is saved with the gyroid infill pattern.

[UMS5_TestFile_GitHub.zip](https://github.com/Ultimaker/Cura/files/4438736/UMS5_TestFile_GitHub.zip)

CTotten-Sonalysts, 2020-04-06:
I could send the fixture file, which is smaller, and doesn't take as long (took about 35 hours).
Or, in reality, you can make your own test file.  Just make a big block, that will have lots of infill.  When I have time to watch prints in progress, layers without infill (like the bottom layers) seem to track with the clock.  Layers with infill, particularly lots of infill, start adding time to the process.

Ghostkeeper, 2020-04-08:
I think I know why this is. It's got to do with the high resolution of the Gyroid pattern and the fast printing speed. It's causing buffer underruns in your printer firmware.

The printer firmware pre-processes your g-code. It'll keep buffers of parsed g-code in various stages, so that if the processor is too slow to keep up with the movement of the print head, the print head has a few more moves that it can go through without needing much attention from the processor. If there are long (low-resolution) moves afterwards, the processor has time to catch up again then.

If this buffer is running out, then the print head will slow down to allow the processor to catch up. In the case of the S5 it's probably not even the motion planner needing to catch up but the main Linux board where it gets the commands from.

There is no way in which Cura can predict the print head slowing down for lack of processing power though. I don't see how we can improve here.

smartavionics, 2020-04-08:
I did submit a PR that included a setting to have lower resolution versions of the TPMS infill patterns but you rejected it.

CTotten-Sonalysts, 2020-04-08:
So you are saying that the Ultimaker S5 is not capable of keeping up?  If this is true, I am assuming the previous Ultimaker machines will also have this issue.  Then why is this infill pattern included in the Ultimaker sponsored slicer program?

Ghostkeeper, 2020-04-14:
> I did submit a PR that included a setting to have lower resolution versions of the TPMS infill patterns but you rejected it.

You closed this PR, probably because we took too long. It took a long time for us to get to reviewing this because it was a rather big change, all rolled into one commit of 1700 lines of code, with a refactor in it, 2 new infill patterns, vertical scaling for infill and separate resolution settings for infill patterns. Nobody dared to take on that change as an in-between task, and our PO couldn't explain why we'd need it so it wasn't scheduled in either.

> So you are saying that the Ultimaker S5 is not capable of keeping up? If this is true, I am assuming the previous Ultimaker machines will also have this issue. Then why is this infill pattern included in the Ultimaker sponsored slicer program?

Yeah. For the most part it won't actually do harm to your print aside from reduced productivity. I'd expect that the Ultimaker 3 does much better in this regard and that firmware version 5.4 also does better than 5.2. I'd expect that the problem pretty much doesn't exist with the Ultimaker 2.

It's included in an Ultimaker-sponsored slicer application because the slicer still allows you to choose settings that will make your print sub-optimal if you go into "Custom" settings. If you'd stick to "Recommended" you'll be fine. Aside from this issue, the Gyroid infill pattern still has advantages too though. I've written some of the properties of Gyroid down here (including that printers may find it hard to keep up) in the Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/infill_pattern.md#gyroid

smartavionics, 2020-04-14:
> You closed this PR

Yeah, you're right there, it wasn't rejected, my mistake.

kdrummer, 2021-02-01:
> Yeah. For the most part it won't actually do harm to your print aside from reduced productivity. I'd expect that the Ultimaker 3 does much better in this regard and that firmware version 5.4 also does better than 5.2. I'd expect that the problem pretty much doesn't exist with the Ultimaker 2.
>
FWIW, I'm seeing this same issue with gyroid infill on the UM2E+