Ultimaker/Cura #7432: Unable to start slicing Cura 4.5

thzpcs, 2020-04-06:
**Application version**
Cura 4.5.0 Windows 64 Bit

**Platform**
Windows 10 x64 
CPU: AMD Ryzen 2700x
RAM: 32 GB DDR4
GPU: Nvidia GTX 1080

**Printer**
Creality Ender 3

**Reproduction steps**
1. Loaded any model into Cura and attempted to slice it, slicing progress remained at 0% indefinitely
2. Deleted all files, reinstalled Cura, problem persists.
3. Installed Cura on another machine, was able to successfully slice. Exported all preferences, loaded on problem machine, still will not slice.

**Screenshot(s)**
[Slicing after 2 minutes](https://user-images.githubusercontent.com/37026570/78565525-0cdcb700-77ec-11ea-9a7e-080870b4f915.png)

**Actual results**
Slicing remains stalled at 0%

**Expected results**
Slicing should commence

**Project file**
Any STL

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4438733/cura.log)

nallath, 2020-04-06:
It seems that Cura was unable to connect with the backend. Do you have some firewalls running? They have been known to block the (local) connection that Cura uses for this.

thzpcs, 2020-04-06:
I have the default Windows firewall setup. Are there specific exceptions that need to be made for Cura?

Ghostkeeper, 2020-04-08:
Not normally. I also have the default Windows stuff on my Windows PC and it's not blocking it.

The reason why we're asking is because Cura communicates to CuraEngine (the slicing back-end) via a local socket. It's a fairly common thing to do on Linux but not so common on Windows, and used sometimes by less savoury applications, so some firewalls prevent this.

thzpcs, 2020-04-14:
I've solved part of the problem. It seems that the issue is with ExpressVPN, which I have on my system. When the VPN is active, Cura will not properly connect to the backend, which is strange as Express is only configured to tunnel specific traffic, so I'm unsure as to why it's effecting Cura

Ghostkeeper, 2020-04-21:
For clarity, CuraEngine will wait on local socket 127.0.0.1:49674, and then Cura's front-end communicates to that socket. If that socket is unavailable, they will try 49675, 49676, etc.

Ghostkeeper, 2020-04-21:
I'm not really sure what we can do about this issue. I don't think we're about to communicate via a different way than libArcus any time soon.

fpdavis, 2020-06-25:
I am having the same exact problem but I don't have a VPN installed. I have disabled my Windows Defender Firewall as well upgraded, uninstalled, and reinstalled Cura to the latest version (4.6.1). Is there any way to test the connection to 127.0.0.1:49674 by hitting it in a web browser/CURL/Etc? Any way to identify for certain what port it is running on?

2020-06-25 15:25:27,011 - DEBUG - [EngineErrorThread] UM.Backend.Backend._backendLog [106]: [Backend] [ERROR] Could not connect to the given address

Ghostkeeper, 2020-07-02:
One thing you can try is to just open CuraEngine.exe in a terminal. Try to execute `C:\Program Files\Ultimaker Cura 4.6\CuraEngine.exe`. It should come up with a help screen, but if the executable has been quarantined or corrupted or something it would not work then either.

Ghostkeeper, 2020-07-02:
The front-end chooses the port that the back-end needs to run on. It's in the start-up parameters. By default it will choose 49674, but if there is a problem with that port it will increment to 49675, 49676, etc. This resolves collisions when you're running multiple instances of Cura at the same time.