Ultimaker/Cura #7434: [4.5] Restoring from Backup does nothing

zing113, 2020-04-06:
**Application version**
4.5

**Platform**
Windows 10

**Printer**
Voron 2.1 (Custom profile)

**Reproduction steps**
1. Fresh install of Cura 4.5 on laptop
2. Initiate restore from backup
3. Cura prompts to restart
4. Click restart
5. Cura closes and never re-opens

**Screenshot(s)**
[image]

**Actual results**
Cura closes and never re-opens by itself. Upon restarting cura manually, no restoration from backup occured.

**Expected results**
Cura open automatically after restarting, with profile and settings restored from backup.

**Project file**

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4439470/cura.log)

**Additional information**
I'm having some issues trying to get a second copy of Cura running on my laptop (already have Cura running fine on my desktop). I downloaded 4.5 to my laptop, clicked through the initial setup screens and selected a random printer profile just to get into the software. Then I logged into my Ultimaker account in order to restore from a backup of my desktop version of the program with all my saved profiles etc., but when i go to restore, I get the message "You will need to restart Cura before your backup is restored. Do you want to close Cura now?". When I click yes, it closes Cura and then does nothing. Sometimes upon re-opening Cura I will have to go through the initial setup screens again, as if my first run through was wiped clean somehow. I've attached a screen shot and my log files. Any ideas?

Ghostkeeper, 2020-04-08:
I tried the back-up functionality and it worked fine for me. On the same computer, at least. I have a couple of questions though:

* Did you create the back-up on Windows as well? Or was it sourced from MacOS or Linux?
* After you restored the back-up and went through the welcome screens again, did it properly back-up some things? I can see in your log that it's upgrading something from Cura 4.4 after you rebooted Cura, so I'm thinking it must've restored something. But maybe the version upgrade goes wrong.
* Did you get a message about the configuration being corrupt?

zing113, 2020-04-08:
1. Everything has be done on windows 10, both laptop and desktop PCs.
2. So I went back to double check, because originally all that i saw was that my custom printer was definitely not listed in my printers list so i assumed the restore didn't work. A partial restore didn't cross my mind. So this morning when i opened cura, i was greeted with the same initial setup screen as if from a fresh install so again i just chose a random printer to get through the screens. But then after checking, i see that everything appeared restored (printer profile, material profiles etc), but i did have a "configuration is corrupt" message as well. I reset cura via that error popup, ran a restore from backup again and everything seems to be fully restored now. Some things to note, cura did not automatically open again after restarting for backup restore, and although i did recently update to 4.5, the backup i was using to restore was created after that update so I would guess it wasn't a translating from 4.4 to 4.5 issue. See attached error screenshot.
[image]

Ghostkeeper, 2020-04-14:
> Some things to note, cura did not automatically open again after restarting for backup restore

Yeah, we still didn't get automatically restarting to work on all platforms. It just closes and you need to manually start it again.

The configuration errors seem to indicate there is something wrong with a "Type A Machines Series 1 2014" printer. I tried adding that printer, making a back-up, resetting to factory defaults and then restoring the back-up. It restored the back-up correctly in my case. No configuration errors and it loads the printer directly when starting Cura.

I still don't know how to debug your error then :( Sorry.

zing113, 2020-04-14:
No worries, like I said, at least it's working now even if we don't know why. Thanks for looking into it.

closed 2020-04-14 by zing113