Ultimaker/Cura #7435: Stacked copies

probonopd, 2020-04-06:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when I have to print many copies of an object, and have to remove prints from the build plate frequently.

**Describe the solution you'd like**

Allow for multiple copies to be stacked over each other with a defined gap so that the copies can break away from each other:

* Stack multiple copies of printable items over each other in z direction
* Each copy is exactly n layers (default = 2) apart from each other
* Check that the bottom surface of the object has the same shape as the top surface, in other words: ensure that the same n layer gap between the objects is there in each location
* And only add supports up to the bottom copy, do not add supports in between the copies

**Describe alternatives you've considered**

Manually arrange objects in the slicer or in the CAD software - doable but cumbersome

**Affected users and/or printers**

Everyone with suitable objects who needs to print large quantities. Use case example: Printing many COVID-19 face shields at once

**Additional context**

[image]

Description of the idea and explanation in this video by Devin Montes from MakeAnything: https://www.youtube.com/watch?v=MHKBvk8IAc0&t=715

nallath, 2020-04-09:
Actually implementing this is going to be rather tricky I think. The example that you give already shows that you can't just use the bounding box (or convex hull) to perform this check. So unless someone is aware of a library that can already do this, I don't think it's very feasible to develop this from scratch on our own, since this would take a lot of time.  By the time we would be able to provide this feature, the need for it is already gone. Part of this is also that we have other requests (also relating to COVID) to implement, so we have to make choices...  

Anyhow, this is a prime candidate for being provided as a plugin, so if someone is up for a nice challenge, we'd love to help out with implementing this. 

That being said; Luban 3d already supports this, so at least for the time being, people can use that instead.

probonopd, 2020-04-10:
Thanks @nallath. Maybe a semi-automatic way would be better than nothing - Cura might do the copying but it would be left up to the user to check the sliced result to ensure a n=2 layer gap between the object in all places, like this:

1. Calculate the bounding box
2. Find one point in the x-y plane on which the object is touching (Cura already needs to do this for puttnig the object onto the build plate surface)
3. At the same x-y location, calculate the maximum z position of the object (is this hard to do?)
4. Copy the object and move it to the calculated z position + n*layer height
5. Repeat step 4 for as many copies as the user has asked for
6. Slice; add supports only up to the first but not the subsequent objects

Would this be more realistic?

nallath, 2020-04-10:
We already have bounding boxes but in the example you gave, that would not work since the correct positioning means that the boxes need to intersect already.

puerlin, 2020-04-10:
I think just a multiple copy with the user specifying the offset and number of copies will suffice. I have made almost 70 of the COVID-19 face masks and found that the gap is critical to getting them to separate just right. 2 layers of gap is way too big (I used about 0.8 of a layer height), so using an integer for layer gaps between parts is not going to provide the required fidelity.

Ghostkeeper, 2020-04-14:
I find that the easiest way to do this sort of thing in Cura is by grouping them up and aligning the groups. In this case if you want like 32 masks stacked, you'd first stack one atop the other with the correct distance, group them, duplicate the group and align the duplicate correctly. Group the two groups, duplicate that, and so on. After 5 times you have the correct number of masks.

We're planning to refactor the algorithm that arranges objects on your build plate. However it most likely still won't place them vertically and won't support custom spacing.