Ultimaker/Cura #7436: [4.6] Crashes on close (when using PPA only?)

thopiekar, 2020-04-06:
**Application version**
Build from master

**Platform**
Ubuntu Focal 20.04 + PPA + Nvidia GFX

**Printer**
Doesn't matter

**Reproduction steps**
1. Open Cura
2. Close Cura

**Screenshot(s)**
There is not much to see.

**Actual results**
Crashes (see log below)

**Expected results**
exit 0 :heart: 

**Project file**
Doesn't matter

**Log file**
[Here you are](https://github.com/Ultimaker/Cura/files/4440482/cura-master-aka-4.6-crash.log.txt)

**Additional information**
Would be great if you could take a look onto it :)

Ghostkeeper, 2020-04-08:
I'm seeing the same when I run from source on my home computer, but not on my work laptop. I think it's got to do with using Qt 5.14 or PyQt 5.14, since the error message is about PyQt.

It doesn't seem to do any harm. The settings are saved fine. But it's a bit annoying for me as well.

pgratz1, 2021-04-06:
Looking though the log files, I'm getting the exact same error and have been since early last year.  FWIW I'm on Kubuntu 20.10.   It is a bit annoying, since Kubuntu always wants to handle the error and that seems to make the system slow for a bit.

thopiekar, 2021-04-06:
Don't face this issue anymore. Many things changed since then. At the moment I'm on Kubuntu 21.04 and haven't seen any of these errors for a long time.
@pgratz1 Yes, that is the difference to AppImages. Whenever a crash is detected, the local bug report tool tries to collect all info possible. That's why it goes crazy on your computer. Are you sure you've got all packages installed and up-to-date?
Can you send a clean log of your installation? Just for confirmation that the problem is identical.

closed 2021-04-06 by thopiekar

pgratz1, 2021-04-06:
Sure, here is a log of the command line output:
file:///home/pgratz/cura.log

jellespijker, 2021-04-06:
This should be fixed with 4.9 beta
With this commit https://github.com/Ultimaker/Uranium/commit/e1a389ffc7fb38ffd4ac9cf9d2a78304e5e692be

pgratz1, 2021-04-06:
Ah great!