Ultimaker/Cura #7437: Can't connect to package database or visit the marketplace website in linux

tausciam, 2020-04-06:
**Application version**
4.5 and 4.6

**Platform**
Arch Linux i5-6500

**Printer**
Ender 3

**Reproduction steps**
Click on marketplace. It will say it cannot connect to the package database. Click to go to the web marketplace. It will say server not found.

**Actual results**
It actually works in Windows....both the marketplace and viewing the web marketplace in firefox. There is no firewall on the linux machine. Both machines are behind the same firewall. I can visit all other websites in linux, but not the webmarketplace. I've tried both in firefox and brave. It says server not found.

tausciam, 2020-04-06:
Sorry...website was having problems. It just started working in linux too

closed 2020-04-06 by tausciam

Ghostkeeper, 2020-04-08:
We're seeing a great spike in usage due to the Corona crisis. Those servers should scale automatically but sometimes they are having some trouble keeping up, it seems.

ChrisTerBeke, 2020-04-08:
We have limits to scaling up/down (and cool-off periods) indeed to prevent a DDOS driving up the infrastructure bill. Occasionally this means that a sudden spike in requests can cause a server error (as your request is being blocked by the load balancer).