Ultimaker/Cura #7438: Final top layer pattern

TTTPOB, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
I love the "Bottom pattern initial layer" feature, which allows me the have a nice look and a dense bottom, with global top/bottom pattern set to zigzag, and the initial pattern set to concentric. But I think for some prints the aesthetic feature of the top layer is also important.
**Describe the solution you'd like**
I want a option which allows the "top pattern final layer" to be set

**Describe alternatives you've considered**
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

**Affected users and/or printers**
I think this would affect all printers

**Additional context**
like this print (ignore the glitches I forgot to balance the plate)

[image]

[image]

Ghostkeeper, 2020-04-08:
This already exists. We call it the "Top Surface". You'd set the [Top Surface Skin Layers](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/shell/roofing_layer_count.md) to 1, and then set the [Top Surface Skin Pattern](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/experimental/roofing_pattern.md) to concentric or whatever you prefer.

closed 2020-04-08 by Ghostkeeper

theQuicker1, 2023-01-11:
It would be nice to have the ability to make the first and last layer to be concentric, not just the top or bottom of the bottom layers. For example, in a vase, it is desirable to have the concentric pattern visible on the top and bottom of the 4 base layers, but not for all 4 base layers because the concentric pattern is structurally weak.

1: Concentric
2: Lines
3: Lines
4: Concentric

A vase has no top layers or top skin so the current setting do not allow the above layering.