Ultimaker/Cura #7439: No brim in tight concave areas

Sophist-UK, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
When you get brim in a small tight concave area it can be difficult to remove and because the area is small the additional adhesion is minimal.

**Describe the solution you'd like**
Cura should avoid putting brim in small concave areas (like drill holes or small indentations in the model.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
All

Ghostkeeper, 2020-04-08:
Like this?
[image]

I did that by using a skirt with 0 distance, but I reduced the "rounding" to 10mm instead of 100mm.

Sophist-UK, 2020-04-08:
I was thinking more like (say) 3/4 of a circle.

Ghostkeeper, 2020-04-14:
I'm not entirely sure what you mean by that? Where should it draw 3/4th of a circle?

Sophist-UK, 2020-04-14:
Apologies for the poor quality diagram...

[image]

If the circle was 100mm across, I would want brim inside it.

If the circle was 2mm or 5mm across I might find it difficult to remove the brim cleanly from inside the hole, and that tiny amount of brim would probably not help with bed adhesion or preventing warping. So instead why not just follow the lines of the walls until they meet and form a bed corner there (as if the circular hole were not part of the object). 

Sorry if it's a poor explanation.

Ghostkeeper, 2020-04-21:
Ah, yeah I understand. I'd implement that with a similar thing as what the skirt does though, albeit a smaller radius. Or maybe we'd do a closing operation with a miter rather than round for the brim. The miter shape is a pretty small detail though.

I'll put this up with my colleagues to see what they feel about it.

Ghostkeeper, 2021-09-07:
We forgot to get back to you on this back then. But this issue got accepted into our backlog. Developers, see CURA-7401.