Ultimaker/Cura #7442: "All at once" print sequencing

Sophist-UK, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
I tried to print 3 objects using "All at once" sequencing and it did the following sequence:

Brim1
Brim2
Brim3
FirstLayer1
FirstLayer2
FirstLayer3
etc.

This resulted in several unnecessary retractions and moves between objects with stringing across objects and was significantly slower than it needed to be.

**Describe the solution you'd like**
IMO Cura should print each layer for one object before switching to another object. It should also try to find a decent point to start printing on the next object which is as close as possible to the end point for the layer on the current object.

Note: The above assumes a single extruder - with multiple extruders I would expect that you might want something like:

Object1 Layer1 Extruder1
Object2 Layer1 Extruder1
Object2 Layer1 Extruder2
Object1 Layer1 Extruder2
Object1 Layer2 Extruder2
Object2 Layer2 Extruder2
Object2 Layer2 Extruder1
Object1 Layer2 Extruder1

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
All

Ghostkeeper, 2020-04-09:
What would you expect to happen in this case?

[image]

I'd expect it with your change to print a communal brim first, then fill in the innermost lines of one part of this brim and print the part within that, then fill in the innermost lines of the other part of the brim and print the part within that.

There's two reasons why it's currently printing like this.

First is that with multi-extrusion, preventing extruder switches is considered the second-highest priority for optimising the order (the very highest priority being that the layers are printed from bottom to top). The case you present above for multi-extrusion is not desired since it introduces multiple switches per layer. This would mean priming multiple times with the prime tower, too. If you're printing 30 dual-extrusion objects you'd need 

Second reason is that we consider switching the flow a greater penalty than making a travel move. Interrupting the flow briefly will change the flow rate out of the nozzle, but according to our tests not as much as switching between the printing speed of the brim vs. the print speed of the infill. This is why Cura will generally print each type of structure separately, rather than e.g. intermix the printing of support with the printing of walls.

Sophist-UK, 2020-04-09:
My proposal does not switch extruders more than once per layer - I quite understand that avoiding changing extruder is 2nd priority over printing layer by layer from the bottom.

What I am saying is that IMO it should print the entire LAYER1 EXTRUDER1 OBJECT1 first before switching to LAYER1 EXTRUDER1 OBJECT2.

That said, if this issue applies only to LAYER1 and it does all other layers as:

LAYER2 EXTRUDER1 OBJECT1 INFILL
LAYER2 EXTRUDER1 OBJECT1 WALLS
LAYER2 EXTRUDER1 OBJECT2 INFILL
LAYER2 EXTRUDER1 OBJECT2 WALLS
LAYER2 EXTRUDER2 OBJECT2 INFILL
LAYER2 EXTRUDER2 OBJECT2 WALLS
LAYER2 EXTRUDER2 OBJECT1 INFILL
LAYER2 EXTRUDER2 OBJECT1 WALLS
LAYER3 EXTRUDER2 OBJECT1 INFILL
LAYER3 EXTRUDER2 OBJECT1 WALLS
LAYER3 EXTRUDER2 OBJECT2 INFILL
LAYER3 EXTRUDER2 OBJECT2 WALLS
LAYER3 EXTRUDER1 OBJECT2 INFILL
LAYER3 EXTRUDER1 OBJECT2 WALLS
LAYER3 EXTRUDER1 OBJECT1 INFILL
LAYER3 EXTRUDER1 OBJECT1 WALLS
etc.

then most of the print will be optimised, and it is probably not worth worrying about optimising first layer time.