Ultimaker/Cura #7444: "One after another" sequencing

Sophist-UK, 2020-04-07:
**Is your feature request related to a problem? Please describe.**
I wanted to print a small object and a tall object one after the other. Cura said it couldn't because the tall object would impact the gantry when printing the short object.

**Describe the solution you'd like**
Cura should not give this error message if there is only one object which is taller than the gantry size, and should print that object last.

**Describe alternatives you've considered**
None.

**Affected users and/or printers**
Anyone printing multiple items One After Another.

Sophist-UK, 2020-04-08:
Just tried to print a single object with One After Another set and got the same message. **Clearly** there can only be a problem with gantry if there is more than one model.

Ghostkeeper, 2020-04-09:
Your actual issue is a duplicate of https://github.com/Ultimaker/Cura/issues/6627.

But when you have only one model loaded, I'd expect that you'd see your build volume be increased to the normal printer height, rather than the gantry height. That's what I'm seeing in any case. Here is a screenshot of one object (which is taller than the gantry height):

[image]

And here is a screenshot of two objects, where the build volume gets reduced to the gantry height:

[image]

Sophist-UK, 2020-04-09:
Yes - it does appear to be a duplicate of #6627.

closed 2020-04-09 by Sophist-UK