Ultimaker/Cura #7445: Cura 4.5 appears top have problem processing JPG files if "Base" is greater than "Height"

CluelessNewB, 2020-04-07:
**Application Version**
4.5

**Platform**
Windows 10

**Printer**
Creality CR-10S 

**Reproduction steps**
There appears to be a bug processing jpg files.    If I load a jpg file and leave all of the settings as the defaults it seems to behave as expected.  If I change  the value for "Base" to something that is larger than the value for "Height" it appears to get really confused.   This works fine in Cura 4.4.1.  This problem appears before slicing.  

 
**Screenshot(s)**
This is with the default settings (height =2.5 base=0.4, darker is higher) it  looks ok:  
[image]

This with height =2.5 and base=20.0, darker is higher) it appears really confused:
[image]
 
This is using Cura 4.4.1 height = 2.5 and base =20.0 darker is higher and appears to be correct:

[image]

**Actual results**
Please see examples above, Cura 4.4.1 processes jpg correctly, Cura 4.5 does not.  This is repeatable with any black and white JPG file.  

**Expected results**
I would expect to see the characters to be raised above the base not below.  

**Project file**
This is the JPG file I have been using but any black and white JPG appears to have the same problem: 
[image]

**Log file**
N/A

**Additional information**
N/A

CluelessNewB, 2020-04-30:
This bug still exists in Cura 4.6.1

Ghostkeeper, 2020-05-06:
Yeah, we didn't get around to it. One of our developers said he'd take a look if he could make a small fix but had to do stuff for the 4.6 release first. And then when that was done I think he forgot this. I'll prod him again.

CluelessNewB, 2021-01-25:
4.8 still has problems importing jpg & png files although behavior is different.  There still seems to be some confusion, it appears to add base ok but the jpg object itself is too high.  For example if I set the base at 20mm and the height as 5mm I would expect the overall height to be base+height (25mm)0.   What I get is 45mm 2 x base + height.

rburema, 2021-02-05:
Seems to be fixed in the current main branch (which is going in 4.9), versus 4.8.

[image]

As such, I'm closing this ticket. Sorry it took so long to resolve.

closed 2021-02-05 by rburema