Ultimaker/Cura #7447: Request for Anti Aliasing

Ben-Boardman, 2020-04-07:
It would be great if the preview window had anti aliasing. It would allow creators to see in better detail despite having a lower resolution monitor (say 1080p or less). I would suggest making it an option i the preferences menu for those who have more capable graphics cards.

Ghostkeeper, 2020-04-14:
We've discussed this with the team and agreed that it would help to make Cura look more modern. We're putting it on our backlog.

alvaro893, 2020-08-20:
I am looking forward to have anti aliasing as well.

T0biasCZe, 2022-02-22:
If you have Nvidia graphics card, you can force anti aliasing in nvidia control panel as a workaround

inputoutput1126, 2022-03-02:
May I point out that SSAA would be most beneficial here since it yeild the most detail.

MariMakes, 2023-05-17:
Quick update from our side 👋

We are cleaning up our open issues, so we can focus on the most requested and needed features and bug fixes from our community and from UltiMaker, I stumbled upon your issue and noticed that it hasn't gotten a proper update.

We've added a ticket to the backlog with the intent to improve this. I just discussed it with the team and they still see value in it. 
For internal reference CURA-7363

Thanks for the report! 👍