Ultimaker/Cura #7448: Rescale doesn't work after a rotation

anthonytrautman, 2020-04-08:
**Application version**
Cura 4.5.0

**Platform**
Win 10 64-bit

**Printer**
Creality Ender -3

**Reproduction steps**
1. Load in a *.STL
2. Rotate model 90deg
3. rescale to 80% in X and Y only (100% in Z uncheck 
4. Y is rescaled automatically to an unexpected number and Z and/or X is scaled up or down 
5. Unable to recover without resetting the model
6. Recently discovered If I scale first then rotate everything works as expected. I believe this is the clue to the bug!

**Screenshot(s)**
step 1
[image]

[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

Ghostkeeper, 2020-04-09:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4136, one of the more common open bugs, but it's pretty hard to solve for us :(

Ghostkeeper, 2020-04-09:
To be precise, it's a duplicate of https://github.com/Ultimaker/Cura/issues/4136, https://github.com/Ultimaker/Cura/issues/3625, https://github.com/Ultimaker/Cura/issues/4309, https://github.com/Ultimaker/Cura/issues/4386, https://github.com/Ultimaker/Cura/issues/5221, https://github.com/Ultimaker/Cura/issues/5189, https://github.com/Ultimaker/Cura/issues/5310, https://github.com/Ultimaker/Cura/issues/5462, https://github.com/Ultimaker/Cura/issues/5599, https://github.com/Ultimaker/Cura/issues/5602, https://github.com/Ultimaker/Cura/issues/5603, https://github.com/Ultimaker/Cura/issues/5872, https://github.com/Ultimaker/Cura/issues/5916, https://github.com/Ultimaker/Cura/issues/5965, https://github.com/Ultimaker/Cura/issues/6148, https://github.com/Ultimaker/Cura/issues/6150, https://github.com/Ultimaker/Cura/issues/6238, https://github.com/Ultimaker/Cura/issues/6370, https://github.com/Ultimaker/Cura/issues/6619, https://github.com/Ultimaker/Cura/issues/6770, https://github.com/Ultimaker/Cura/issues/6829, https://github.com/Ultimaker/Cura/issues/6890, https://github.com/Ultimaker/Cura/issues/6971, https://github.com/Ultimaker/Cura/issues/7102, https://github.com/Ultimaker/Cura/issues/7142, https://github.com/Ultimaker/Cura/issues/7237, https://github.com/Ultimaker/Cura/issues/7388 and https://github.com/Ultimaker/Cura/issues/7446 :rofl:

closed 2020-04-09 by Ghostkeeper