Ultimaker/Cura #7449: Bug slice hole

userosos, 2020-04-08:
Cura 4.5 (Cura 4.6) Windows 10 x64 has the error also. Not correct slice some hole in the model. 
[image]
[CFFFP_mainsail_entry_NEW_long_PRINT_1(3).zip](https://github.com/Ultimaker/Cura/files/4450822/CFFFP_mainsail_entry_NEW_long_PRINT_1.3.zip)

smartavionics, 2020-04-08:
Please provide the project file (File -> Save, rename or zip .3mf) or at lease the STL so we can check the model(s). Thanks.

userosos, 2020-04-08:
> 
> 
> Please provide the project file (File -> Save, rename or zip .3mf) or at lease the STL so we can check the model(s). Thanks.

I checked the models it not work.

smartavionics, 2020-04-08:
> I checked the models it not work.

OK, if you don't want to supply any more info, I am not bothering to respond to this issue any more, goodbye.

userosos, 2020-04-08:
> OK, if you don't want to supply any more info, I am not bothering to respond to this issue any more, goodbye.

I think that i not understend you. I mean that i add 3mf file in the first message. I repeat now again.
[CFFFP_mainsail_entry_NEW_long_PRINT_1(3).zip](https://github.com/Ultimaker/Cura/files/4451458/CFFFP_mainsail_entry_NEW_long_PRINT_1.3.zip)

Ghostkeeper, 2020-04-09:
That's a .gcode file, not the project file (.3mf).

userosos, 2020-04-09:
@Ghostkeeper I apologize for my stupidity
[CFFFP_EPSILON_113_ins_end_simple_part2 GREG PETg lw 0.4mm lh 0.2mm if 30% ext1 210C bed 60C.gcode.zip](https://github.com/Ultimaker/Cura/files/4458142/CFFFP_EPSILON_113_ins_end_simple_part2.GREG.PETg.lw.0.4mm.lh.0.2mm.if.30.ext1.210C.bed.60C.gcode.zip)

eskeyaar, 2020-04-09:
> 
> 
> @Ghostkeeper I apologize for my stupidity
> [CFFFP_EPSILON_113_ins_end_simple_part2 GREG PETg lw 0.4mm lh 0.2mm if 30% ext1 210C bed 60C.gcode.zip](https://github.com/Ultimaker/Cura/files/4458142/CFFFP_EPSILON_113_ins_end_simple_part2.GREG.PETg.lw.0.4mm.lh.0.2mm.if.30.ext1.210C.bed.60C.gcode.zip)

still its a gcode, not a 3mf project file.

userosos, 2020-04-09:
> still its a gcode, not a 3mf project file.

Oh sorry. I think that i zipped 3mf file.
[CFFFP_EPSILON_113_ins_end_simple_part2.zip](https://github.com/Ultimaker/Cura/files/4458552/CFFFP_EPSILON_113_ins_end_simple_part2.zip)

eskeyaar, 2020-04-10:
> Oh sorry. I think that i zipped 3mf file.
> [CFFFP_EPSILON_113_ins_end_simple_part2.zip](https://github.com/Ultimaker/Cura/files/4458552/CFFFP_EPSILON_113_ins_end_simple_part2.zip)

The model looks good only. what issue are you facing? you mean the hole is small? if so then change in desgin. Everything looks ok in the preview.

userosos, 2020-04-10:
> 
> 
> > > > @Ghostkeeper I apologize for my stupidity
> > > > [CFFFP_EPSILON_113_ins_end_simple_part2 GREG PETg lw 0.4mm lh 0.2mm if 30% ext1 210C bed 60C.gcode.zip](https://github.com/Ultimaker/Cura/files/4458142/CFFFP_EPSILON_113_ins_end_simple_part2.GREG.PETg.lw.0.4mm.lh.0.2mm.if.30.ext1.210C.bed.60C.gcode.zip)
> > > 
> > > 
> > > still its a gcode, not a 3mf project file.
> > 
> > 
> > Oh sorry. I think that i zipped 3mf file.
> > [CFFFP_EPSILON_113_ins_end_simple_part2.zip](https://github.com/Ultimaker/Cura/files/4458552/CFFFP_EPSILON_113_ins_end_simple_part2.zip)
> 
> The model looks good only. what issue are you facing? Do you mean the hole is small? if so then change in design. Everything looks ok in the preview.
> [image]
May be. Prusaslicer work is Ok. The model not from me. I checked the mode

Ghostkeeper, 2020-04-14:
The model does have a hole there.
[image]

What's wrong with the hole, exactly? Could you fill in the issue template? What is the expected behaviour and what is the behaviour that you're seeing?

closed 2020-06-19 by no-response[bot]