Ultimaker/Cura #7450: Custom start Gcode included twice

Sophist-UK, 2020-04-08:
**Application version**
4.5

**Platform**
Windows

**Printer**
Dagoma DiscoEasy200

**Reproduction steps**
Put custom Start Gcode in the machine definition.
Saved the Gcode for an object to file.
Examined the file.

**Actual results**
Custom Gcode inserted twice, first preceded by `M82 ;absolute extrusion mode`, second preceded by `G92 E0`.

**Expected results**
Custom start Gcode should only be included once.

Sophist-UK, 2020-04-08:
Interestingly, after I removed the custom code in the machine definition in Cura settings, the Gcode contained one copy of the Gcode and zero copies of the default Gcode.

Ghostkeeper, 2020-04-14:
Hmm, I'm not getting the same result. I see it only once:

```
;FLAVOR:Marlin
;TIME:634
;Filament used: 0.238755m, 0m
;Layer height: 0.1
;MINX:93.275
;MINY:93.275
;MINZ:0.26
;MAXX:111.725
;MAXY:111.725
;MAXZ:9.96
;Generated with Cura_SteamEngine master
T0
M104 S205
M105
M109 S205
M82 ;absolute extrusion mode
;My custom start g-code
G92 E0

;Start T0
G92 E0
G1 E-3.5 F10000
G92 E0G1 E1.5 F3000
G1 E-60 F10000
G92 E0

G92 E0
G1 F3000 E-3.5
;LAYER_COUNT:98
;LAYER:0
M107
G0 F1285.7 X107.536 Y94.335 Z0.26
;TYPE:SKIRT
G1 F3000 E0
G1 F900 X108.038 Y94.382 E0.0193
G1 X108.525 Y94.508 E0.03855
G1 X108.987 Y94.71 E0.05784
```

Are you sure you didn't also paste it in the extruder start g-code?

Sophist-UK, 2020-04-14:
It is possible. I deleted the printer in order to correct it. But I will try to recreate the error, and if I am unable to do so, I will close this issue.

closed 2020-04-24 by Sophist-UK