Ultimaker/Cura #7452: Incomplete Backups

yeowjoon99, 2020-04-09:
**Application version**
All

**Platform**
Windows 10

**Printer**
Doesn't matter

**Reproduction steps**
1. Backup data
2. Uninstall old version of Cura
3. Install latest version of Cura
4. Restore data

**Screenshot(s)**
-NIL-

**Actual results**
Not everything is restored. Essential data like the filament cost and weight of all my materials
And small details like preferences in the "Configure Cura..."
This means that with every single Cura update, I lose the cost and weight of every single filament I've ever used

**Expected results**
All parameters such as filament cost and filament weight should be backed up and restored
The settings in "Configure Cura..." would ideally be backed up as well but isn't as essential as the pricing and weight of the filaments

**Project file**
-NIL-

**Log file**
-NIL-

**Additional information**
-NIL-

Ghostkeeper, 2020-04-14:
Hmm, not able to reproduce this issue here. These were my reproduction steps:
* Hide the Layer Height setting.
* In the Material Manager, change the weight of Generic PLA to 750g and the cost to 30 euros.
* Create a back-up.
* Close Cura and do a factory reset.
* Open Cura, log in and restore the back-up.

Now the layer height setting is not visible, since the setting visibility was restored. The material weight and cost are also restored correctly.

Maybe the issue is only on Windows?

The things that you name happen to all be stored in the preferences file (cura.cfg). They are not part of any profiles. Maybe that is a hint as to where this bug is coming from.

yeowjoon99, 2020-04-14:
@Ghostkeeper I understand that the parameters in the settings would be stored in the config file but what's data like filament weight and price doing in there 🤔

Also where's the cura.cfg file stored, maybe it would help if I could put it here

Ghostkeeper, 2020-04-14:
> what's data like filament weight and price doing in there

Yeah, a bit weird indeed, but it's the result of our distribution. The profiles for the materials that come pre-shipped with Cura are stored in Cura's installation folder. That folder is write-protected (e.g. on Windows it's in Program Files) so Cura is unable to modify those files. Since we still wanted people to be able to write down prices and weights for built-in materials we opted to store them in the preferences instead.

konskarm, 2020-04-17:
Hi @yeowjoon99. I tried reproducing this issue on Windows, but I did not manage.
I also changed the visibility of some settings and the cost/weight of a few materials, made a backup, removed all the previous configurations, run the new version and restore the backup. I am getting all the settings back without an issue. 
Can you try reproducing it once again?

yeowjoon99, 2020-04-17:
I've actually had this problem for quite a few versions now, I just didn't upload it as an issue till now.

It's happened to me with literally every update for as long as I could remember.

I haven't tried it on a different PC (cause I don't have one) but based on what both of you are saying, it could be an issue with the backing up or restoring process.

Another thing that could be affecting this is the fact that my machine is "Custom FDM Machine" which has caused other problems in my experience