Ultimaker/Cura #7456: [4.4+] Clearing out "Search settings" is very slow

szopen111, 2020-04-09:
**Application version**
4.4, 4.5, 4.6beta. Didn't check earlier versions.

**Platform**
(all x64) Windows 7 + NVIDIA GTX 950, Windows 10 + embedded Intel 
[slow_search_settings.log]

GPU, Ubuntu 18.04, Linux Mint (virtualized)

**Printer**
(shouldn't matter) Ender-3 

**Reproduction steps**
- open the slicer settings and choose "all"
- expand every group (Quality, Shell, Infill, ...)
- type in "ada" in the "Search string"
- the "Use Adaptative Layers" shows up (good)
- now click the "X" icon. Making all the options visible takes several seconds (see the video)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
Restoring the visibility of all settings is very slow.

**Expected results**
Restoring the visibility of all settings should be instantaneous.

**Project file**
n/a (empty project)

**Log file**
Only startup logs. Filtering doesn't add anything to the logs.
(https://github.com/Ultimaker/Cura/files/4458086/slow_search_settings.log)

**Additional information**
n/a

szopen111, 2020-04-09:
I've tested this with default settings (renamed the `%APPDATA%\cura` folder). Same issue.

[image]

fieldOfView, 2020-04-10:
Cura is somewhat stuck between a rock and a hard place here. There are *lots* of settings, and you have chosen to make them all visible. I order to make the scrolling go as smoothly as possible, Cura is configured to draw all setting controls into a "graphical cache". Creating this caches takes time and unfortunately the cache gets invalidated when changing the visibility of all those controls.

The best current workaround is to not make all settings visible. Be critical of the set of settings you actually need and use the rest via the search.

Ghostkeeper, 2020-04-14:
A significant redesign of the settings list is probably the only way to get rid of this. We did make some good improvements in this regard in Cura 4.4 and a bit more in 4.5.

fieldOfView, 2023-05-30:
You may want to try this nightly build from a little while ago; it performs better for me.
https://github.com/Ultimaker/Cura/discussions/15098
The major change in that build will be part of Cura 5.4.

MariMakes, 2023-05-30:
Hey @DamianSepczuk,

@fieldOfView is right. There are some speed improvements coming in Cura 5.4.

Can you please try it out and let us know if it resolves your issue?
You can find the download links and the discussion here: https://github.com/Ultimaker/Cura/discussions/15098

DamianSepczuk, 2023-05-30:
Hi! Thanks for getting back to me :) Since reporting the bug I've upgraded both my workstation and Cura, so it's hard to compare.
The responsiveness difference between 5.2.2 and the nightly build from #15098 is negligible. Both are usable, but the list loading, after removing the filter, is still somehow laggy.

From removing the filter "ada" to having the unfiltered list rendered it took:
 - **5.2.2**: ~1.2s
 - **release_15098**: ~1.14s

It might be worth testing this on an older hardware.

## release_15098
[image]

## cura_5 2 2
[image]

MariMakes, 2023-06-05:
Hey @DamianSepczuk,

Thanks for the nice gifs. 🤩 
I'm also glad to hear that you see some improvement over what you had before (and have upgraded your hardware 😉)

As @fieldOfView and @Ghostkeeper mentioned before, it's a careful balancing act and there are already some improvements and workarounds in place. 

We believe we will not work on this anytime soon and will therefore close this issue. 
I hope you understand.

closed 2023-06-05 by MariMakes