Ultimaker/Cura #7457: Network connection lost to Ultimaker s5 Pro Bundle

jficaraharvard, 2020-04-09:
**Application version**
(The version of the application this issue occurs with.)
4.5.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)
Windows 10 Version 1909 (OS Build 18363.752)

**Printer**
(Which printer was selected in Cura?)
Ultimaker s5 with Pro Bundle (Both air handler and material station)

**Reproduction steps**
1. Loaded a model
2. Sliced model
3. Sent model to Ultimaker s5 over the hard wired network connection

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
Cura Version screen shot
[image]

After printing for about 30 minutes or so, sometimes more sometimes less
I see this screen:
[image]

[image]
[image]
[image]

**Actual results**
(What happens after the above steps have been followed.)
I loose connection to the printer

**Expected results**
(What should happen after the above steps have been followed.)
I don't loose connection to the printer

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)
This is not specific to any paticular model

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
[cura.log](https://github.com/Ultimaker/Cura/files/4458951/cura.log)

**Additional information**
(Extra information relevant to the issue.)
I tried rebooting my PC and the network connection is still not available to the Ultimaker s5 from Cura
I tried performing a Cura reset from the ultimaker s5 and that clears the issue, meaning Cura is able to connect to the Ultimaker s5 again, but the issue, loss of network connection, comes back after the next print.

Turning off the Ultimaker s5 and turning it back on again also resolves the issue, in that the network connection comes back until a new print is sent and prints for about 30 minutes or so.

I verified, when the network connection was lost, that the ultimaker s5 is online both from my router and from the ultimaker s5 front panel display

I filed an issue with Ultimaker support but they indicated it was a Cura issue and asked me to file a bug report here.

Note the logfile I uploaded today 2020-04-09 may have some log information in it from today 2020-04-09 with no connection to the Ultimaker s5 but that is because its off at the moment.

nallath, 2020-04-10:
Have you tried adding the machine directly by IP? We've noticed that some networks have issues with the automatic detection system.

jficaraharvard, 2020-04-10:
Yes I have tried adding the machine by IP address directly.

The issue still occurs where Cura looses connection to the machine after printing starts and runs for a few minutes.

Best regards,

Joseph Ficara

nallath, 2020-04-10:
Alright, thanks for confirming that. I've checked the logs that you provided, but it doesn't seem to be an issue on the side of Cura. The printer itself is replying that the Service is unavailable (So it's unable to answer the request that we are sending to it).

That would also explain why the printer is still online in your router (because it is). It also explains why turning the printer off restores the functionality for a bit, since then the service that we're asking for the info we need is rebooted again.

So, I'm very sorry, but I don't think there is much that we can do about this from the side of Cura.

Ghostkeeper, 2020-04-14:
Can confirm, we've seen this happening as well since the 5.4 update of the firmware on the S5. Ultimaker's System Integration Testing team is up in hairs about this but there is no solution yet afaik. It's pretty hard to reproduce as it only happens on some printers.

The problem is that sometimes some service running on the printer gets stuck in a boot loop. This then takes so much processing power that basically all services on the printer are suspended until processing power is available. The printing service gets higher CPU priority so that is unaffected, but other things like the API that Cura talks to basically won't respond any more. You also won't be able to visit the IP address in a browser any more.

I'll close this since the it's not a bug in Cura.

closed 2020-04-14 by Ghostkeeper