Ultimaker/Cura #7458: [4.6.0 Beta] Most Z Moves Missing

CCS86, 2020-04-10:
4.6.0 Beta / Win 10

Well, this beta version just tore my print head up. After starting the initial layer at 0.2mm, there are no Z moves until like Z11.xx . So, the printer just kept printing at Z.2 until bad things happened :(

[4.6.0.gcode.txt](https://github.com/Ultimaker/Cura/files/4459365/4.6.0.gcode.txt)

smartavionics, 2020-04-10:
That gcode has been post-processed so maybe that's what zapped z? What does the post-processing script do?

CCS86, 2020-04-10:
@smartavionics I think you are right. I had the "retract continue" PP script on, which I didn't connect to Z move modifications, since it should explicitly be creating E moves, not modifying any other axis. Removing that script appears to fix the issue.

I'll link to this issue in the script update issue.

rburema, 2020-04-12:
The fix linked in that ticket was not the only one, did the updated script you donwloaded contain all three fixes? 

(https://github.com/Ultimaker/Cura/commit/d5166030d76838e4aa52c25338081468c2535195, https://github.com/Ultimaker/Cura/commit/9b997d421b25c1bfc56be7c68ba75fd8f6902653 and https://github.com/Ultimaker/Cura/commit/d9649dc3ddd7e055bc936030ae5f9ddc4b20bc7b)

For others following along: Those where added to the 4.6 branch _after_ the beta came out, so the issues are still in 4.6-beta, but _should_ be fixed in 4.6

CCS86, 2020-04-12:
Maybe not. I downloaded a copy of the script linked by ghostkeeper.

rburema, 2020-04-12:
> Maybe not. I downloaded a copy of the script linked by ghostkeeper.

If you mean the one in the issue, then that's the file with only one or two of the fixes applied, not all three.

Ghostkeeper, 2020-04-14:
I don't think the other two fixes would make any difference in this regard. one is a transparent change to reduce indentation. The other is to counter rounding errors that would cause the script to have a false positive (it thinks there is a retraction but actually there isn't).

I could reproduce your issue after a few educated guesses. The culprit is that "Retract on Layer Change" is enabled. The script doesn't deal with that correctly. It also needs to start tracking Z moves to correct this.

Ghostkeeper, 2020-04-15:
Fixed for 4.6 stable.

commit 9f383749 (2020-04-15): Also track Z coordinates and retract extra for the distance travelled then

Otherwise if a retracted move had a different Z coordinate, this would not be put in the final print. It's pretty simple to extend it this way.

Fixes #7458.
closed 2020-04-15 by Ghostkeeper