Ultimaker/Cura #7459: Cura GUI Graphical Issues

Linkeb3, 2020-04-10:
**Application version**
(The version of the application this issue occurs with.)
4.5.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

Manjaro

**Printer**
(Which printer was selected in Cura?)
Ender3

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)
Open cura

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)
the GUI works?

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)
won't open

**Additional information**
(Extra information relevant to the issue.)
This problem is on the AUR as well.

nallath, 2020-04-10:
Did you use the appimage?

Linkeb3, 2020-04-10:
I did not, I’m unfamiliar with appimage.

nallath, 2020-04-10:
Well, the packages provided by some Linux distros have different versions of Qt (the graphical framework that we use). This is known to cause all kinds of issues, including the one that you seem to have. 

Since we don't maintain these packages (and only the appimage) we only give support for the appimage.

Linkeb3, 2020-04-10:
I’ll look into it, thanks for the reply

rburema, 2020-04-12:
The appimage can be downloaded from [here](https://github.com/Ultimaker/Cura/releases). (Scroll down if you want a final instead of a beta.)

MaxVerevkin, 2020-04-16:
try this
env -u DESKTOP_SESSION -u KDE_FULL_SESSION -u XDG_CURRENT_DESKTOP /bin/cura

closed 2020-06-19 by no-response[bot]