Ultimaker/Cura #7461: Disable cura control extruder temp for last layer

SpectroMaledikto, 2020-04-10:
Can I disable all controls temp from Cura ?

[my status]
I have 1 hotend and 2 filaments in one hotend.
The resistor hotend is connected to EXTR 1

[problem]
if the last layer print, is for extruder 2, cura will turn off temp my extruder 1
but I printing with only one extruder 1.

in this case cura will turn off my extruder 1 and aborting print.

what I can do for disable cura temp control extruders for the last layer ?

smartavionics, 2020-04-10:
> what I can do for disable cura temp control extruders for the last layer ?

Recent Cura releases have a setting that says that the extruders share a heater. Enable that and it will solve your problem. The setting is in the printer settings dialog.

SpectroMaledikto, 2020-04-10:
> 
> 
> > what I can do for disable cura temp control extruders for the last layer ?
> 
> Recent Cura releases have a setting that says that the extruders share a heater. Enable that and it will solve your problem. The setting is in the printer settings dialog.

thank you

closed 2020-04-10 by SpectroMaledikto