Ultimaker/Cura #7462: Feature request: Different orbit implementation to avoid right click menu

Bjack795, 2020-04-10:
**Application version**
Cura 4.5

**Platform**
Windows 64

**Printer**
Ender 3

**Issue**
I waited for some versions to fix this issue before opening it here.
I use a lot of 3D modelling softwares and there are a lot of ways of implementing the "orbit".
Using right click is just not good. the 60% of the time when I try to orbit the view I trigger the menu of the right click, this results also in unwanted transform of the model, sometimes I'm deleting it etc...
Please use as other softwares shift+left click, alt+scroll, or whatever.
Just stop using the right click as orbit.

Ghostkeeper, 2020-04-21:
You shouldn't get the context menu unless you release the right mouse button. Are you getting the context menu before you release the button? You might need to look at cleaning the clicker in the mouse maybe. Dirty mice are commonly the cause for buttons being clicked twice.

We have some plans to be able to swap out controls to be the same as other software people may be familiar with. That would also be a solution to this problem and the underlying cause that made you post this problem here.

Bjack795, 2020-04-21:
> You shouldn't get the context menu unless you release the right mouse button. Are you getting the context menu before you release the button? You might need to look at cleaning the clicker in the [...]

I use thousand of 3D modelling softwares and my mouse (my mices) doesn't have any problem with the orbit. Maybe, just maybe changing the shortcut could help also the people with consumed mices not getting mad.

Ghostkeeper, 2020-04-22:
You'd still have the problem with the left mouse button also, mind you, if the left mouse button is bad. The Blender way of using the middle mouse button is the only cure.

autotechmusic, 2020-04-26:
Perhaps the FreeCAD way would be a good option:

Alt + Mouse movement = Rotate Surface
Shift + Mouse Movement = Pan Surface

No Clicking needed and helps out Trackpad and Laptop Users as well.

Bjack795, 2020-04-26:
I think that the way of orbit and panning of the 90% of the softwares could be taken also in cura.
So middle mouse + shift -> orbit, middle mouse only -> panning
Then of course you could add some way to adjust the shortcuts for the trackpad etc.

The best way should be to let to the user the choice of out to navigate according to the comfort of each device.