Ultimaker/Cura #7463: [4.5.0] Cura randomly skips inner walls in thin cylinder

XDA-Bam, 2020-04-10:
**Application version**
Cura 4.5.0

**Platform**
Windows 10 Pro 64bit 1909
Xeon E3-1230 v2
16 GB RAM
NVidia GTX1060 6GB

**Printer**
Ender-3 Pro

**Reproduction steps**
1. Load model of cylinder with central hole
2. Select profile "Standard Quality"
3. Deselect "Fill Gaps Between Walls" (Optional. This makes the problem more obvious. Using "Fill Gaps", the result will be garbled tiny segments instead of a missing wall in those layers. Printed parts, however, look similar and are missing material in affected layers.)
4. Slice
5. Switch to preview, look at layer 27 (or others)

**Screenshot(s)**
Without "Fill Gaps...":
[image]

With "Fill Gaps...":
[image]

**Actual results**
Cura randomly skips inner walls in about 10% of all layers. Others layers look fine with 3 intact walls.

**Expected results**
Cura should slice all layers identical and not skip some internal walls. Cylinder and hole have constant diameters and there is no reason layer 27 should be different from layer 40 or 50.

**Project file**
[CE3_Cylinder.zip](https://github.com/Ultimaker/Cura/files/4462075/CE3_Cylinder.zip)

**Log file**
[cura.log.zip](https://github.com/Ultimaker/Cura/files/4462089/cura.log.zip)

**Additional information**
The issue goes away when changing the line width from 0.4 to (i.e.) 0.3999:
[image]

rburema, 2020-04-12:
I can't seem to reproduce this?

Here's what I see if I load your project into 4.5.0:
[image]

XDA-Bam, 2020-04-12:
@rburema But you did reproduce it. Exactly. Look at the inner wall (green): It's missing in those layers around 27. It's in your screenshot 😄 

The second and third screenshots I provided show different changes: The second is with "Fill Gaps Between Walls" activated to show that there is some garbled infill added instead of a wall. The third is using a different wall line thickness to give you more info on where the problem might originate from.

rburema, 2020-04-12:
I see, I was focusing on the z-seam, which doesn't show up in my image. I have the excuse that it's getting late here though... :-D

When working with a space (between the two walls) where exactly one wall would fit, we're starting do deal with rounding issues: Since the model is made from polygons that can't reproduce a circle _exactly_, the space may technically be too small for a full third wall.

We're eventually going to deal with these sorts of issues by merging [libArachne](https://github.com/Ultimaker/libArachne) into CuraEngine.

XDA-Bam, 2020-04-13:
OK. I already suspected that this might be  a numerical issue. Just wanted to report it to make sure it''s nothing else. Good to know a possible fix is on the way. Closing this for now.

closed 2020-04-13 by XDA-Bam