Ultimaker/Cura #7465: cura 4.5.0 change fillament bucle and don´t print

phiphe, 2020-04-10:
I have a creality CR-10S, i choose this machine in cura.
I generate the gcode, and when i try print in my cr-10s....when i get temperatura to begin, appears "CHANGE FILLAMENT" and always is a bucle ...and not print.
If i use S3D , i use this script and print ok the same stl.

M75 ; Start the print job
G28 ; home all axes
G29 ; bltouch
G1 Z5 F3000 ; lift
G1 X60 Y3 F1500 ; avoid binder clips
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X120 E10 F600 ; prime nozzle
G1 X150 F5000 ; quick wipe 

I can´t use CURA 4.5.0

Can you help me?, i dont have fillament sensor.... and i use marlin 1.1.9 with bltouch and bondtech.
Regards

rburema, 2020-04-12:
It might be a printer issue rather than a Cura one (yes, even tough it prints fine for the other slicer). I saw online that people had issues with that error cropping up 'randomly', due to a faulty wire from what I understood.

That said, could you send the gcode produced by Cura (and maybe also the other slicer)? _Maybe_ we can see what goes wrong.

Also, you can change the start gcode should that be required in the machine settings.

phiphe, 2020-04-12:
Hi,
I tried with sd and usb,......if i try with s3d all ok, if i try with
cura.....change fillament bucle and dont print.
i tried changing script in cura (the same that Works in S3D) and nothing.
i tried with default script in cura and nothing…..
I would like to use Cura, but is imposible.

I have a CR-10S with bltouch and bondtech.

Can you help me?
Regards

phiphe, 2020-04-13:
hi,
Can you send me an email that i could send you gcode?.
This mail returned, and tell me that 2 archives are too bigger.

Regards

rburema, 2020-04-14:
You can send it to `r.burema` `ultimaker` `com`.

rburema, 2020-04-16:
I've received the files. I see you use some custom start gcode? At least it's not the start gcode that gets added whit an unaltered cr10s definition.

phiphe, 2020-04-16:
Yes, i have bltouch,
But with default script i have the same.....

phiphe, 2020-04-16:
I tried too with sd and with USB.
In S3D not problems...but in cura always change fillament bucle....

Ghostkeeper, 2020-04-21:
We can't really help you without access to your printer here. The default g-code contains nothing that should cause the printer to change filament. We can change something if you know which g-code command is the problem, but otherwise we're guessing in the blind here, with millions of users watching us make mistakes that cause their prints to fail.

phiphe, 2020-04-21:
Thanks, is strange because the same code works in s3d but not in cura.
Pherhaps the problem is the default printer that i choose in cura
....cr-10s...i have the board 2.1 .
Continues with s3d...perhaps someday cura....

Regards.

closed 2020-04-21 by Ghostkeeper