Ultimaker/Cura #7467: Cura 4.5 Bug. Per model settings apply to walls

Wdfitzpat007, 2020-04-10:
Vura 4.5

All Operating systems

All Printers

**Reproduction steps**
1: Attempt to use modify settings of overlap, Do not overlap the entire model you want to edit
2: Slice
3: Preview
4: Take blood pressure medication

**Actual results**
Model is cut into separate pieces.

**Expected results**
Only selected settings should be split. Have seen many bug reports reporting the same issue, but none of them address the expected results. The real expected results are: only the modified settings are split. Run each of the slicing steps normally, except the ones the user modified, and only run those settings with the cut model.

Maybe you already thought of that, decided it was not worth your time, or got frustrated, and gave up. Deciding it was easier to name it "Cutting Mesh *of frustration*" and be done with it. Unfortunately I am a print farmer and am tired of my several thousand dollar machines rattling back and like expensive windup toys. It is always caused by the top/bottom pattern not being optimized for the area it is printing in. If you have long segments, concentric is 1,000 times better than lines. If you have a big open area, lines are 1000 times better.

Maybe the easier thing to do is make the slicer automatically notice when it is making a bunch of short moves and try to switch top/bottom pattern in that area to lower the number of moves. Sounds a lot easier to code and would take a lot of work off the user from doing it manually.

Included a sliced model for a low end Prusa mk3. 
[FuNoTake3mfFile.zip](https://github.com/Ultimaker/Cura/files/4463731/FuNoTake3mfFile.zip)

**Project file**

Wdfitzpat007, 2020-04-11:
After sleeping on it, and reviewing my prints, the best interface with infill always comes from concentric supports unless you print an extra wall in the infill. For fast and light prints, this has drawbacks ex: vibration on short moves. 

-Printing two concentric top layers rarely results in a smooth top surface

-Ironing takes too much time

-Fill in gaps does not work for fast print settings

Instead, having the first top layer print concentric then having the next layer print lines minimizes rattling, print time, and maximizes top surface quality. Furthermore it is easily a viable implementation vs having the slicer refactor the top surface into segments as this would be both time expensive to code and CPU time expensive to slice.

mahtDFR, 2020-04-14:
We discussed this in our morning stand up and decided we can't prioritise it right now, so I will defer it. We're open to pull requests if someone wants to work on it.