Ultimaker/Cura #7468: Feature Request: Ability to save notes against each setting

CthulhuStig, 2020-04-11:
First off, I apologise if this is already a existing feature that I just haven't found or if it has already been requested.

I think it would be useful to have the ability to save free form text notes for each setting. When setting up a profile there are generally a lot of settings that get tweaked to get it spot on. Over time it can often be hard to remember details of how specific values were arrived at.

The ability to save notes for settings would allow you to save details of testing that was done. It could also allow you to save notes of alternative values for other use cases (e.g. different temperatures for different filaments). This would prevent the need for creating multiple profiles that are 99% the same and the maintenance problem that entails when you want to make the same tweak across them all.

The obvious alternative would be to save notes in an external document. The advantage of storing them as part of the profile is they are then included if the profile is distributed.

I can imagine a simple notes facility would be of use to may of your users.

Thanks,
John

Ghostkeeper, 2020-04-21:
We've had feature requests of storing notes with a profile before. Not about storing notes with a setting override though. Maybe something can be put in the context menu there or something.

Yian-Lai, 2020-04-22:
We've discussed with the team and put this on our backlog. 
For internal see CURA-7387.

fvrmr, 2021-06-01:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.