Ultimaker/Cura #7469: [4.5-4.6] Weird Support generation

chaoticchoas, 2020-04-11:
**Application version**
4.5 - 4.6 Beta

**Platform**
Windows 10 
Rx 480

**Printer**
Ender 3
Monoprice select mini v2

**Reproduction steps**
Turn on Supports

**Screenshot(s)**
https://imgur.com/a/7MJMPEL

**Actual results**
Weird support generation with gaps in the supports and support floating in the air.

**Expected results**
Supports without any weird gaps or supports floating in the air or randomly ending

**Project file**
[Inner_Shell_Bottom_3mm.zip](https://github.com/Ultimaker/Cura/files/4463942/Inner_Shell_Bottom_3mm.zip)

**Log file**
[cura.log](https://github.com/Ultimaker/Cura/files/4463944/cura.log)

**Additional information**
Not sure if it's relevant I made some edits to the usbprinting plugin. [Here ](https://community.ultimaker.com/topic/30043-does-not-connect-with-ender3-and-skr-mini-e3-board/)is a post of the edits I made.

It also seems to affect conical support generation in which both my printers will have different support generation with most setting be the same except layer height but have vastly different support generation. The strange thing is that in different models it work fine for one printer while causing issues in the other printer.

[Here ](https://imgur.com/a/jwDKlyS) are some examples of what I mean.

rburema, 2020-04-12:
The area for which support is to be generated is very thin. You could try to adjust the 'Support Horizontal Expansion'.

chaoticchoas, 2020-04-13:
Thank you @rburema. I was able to generate the supports without any issues. I just now need to figure out what is causing my issues with conical support generation. Thanks for your help.

closed 2020-04-13 by chaoticchoas