Ultimaker/Cura #7470: M106 lerdge k

valaleks, 2020-04-11:
GCode in Cura for Lerdge K. I have 2 fans, there is a need for separate control, but Cura is built in GCode in the format: M106 S255 P1; (blower fan control model for FAN1). But Lerge perceives this command for both fans at the same time, i.e. both begin to spin. Lerdge controls the fan through a command (from the Gcode list): M106 T1 S255;

Ghostkeeper, 2020-04-21:
You can do this with a Search and Replace post-processing script. For instance you could replace all occurrences of `M106 S255 P1` with `M106 T1 S255`. Or more advanced maybe but working for all fan numbers and fan speeds, something like `M106 S(\d+) P(\d+)` with `M106 T\2 S\1` using regex. I didn't test this, but I hope you can try if that works for your firmware.

We don't really have a way to implement this for you since we can't verify that this works for your printer. So we're going to have to deny this, unless you implement this yourself. Preferably this sort of thing would be implemented with a machine setting that determines the format for fan speed commands.

closed 2020-04-21 by Ghostkeeper

valaleks, 2020-04-22:
Спасибо! Мне помогло это решение

Ghostkeeper, 2020-04-28:
Graag gedaan, makker.

FranciscoBellaG, 2023-12-02:
busco perfiles de cura para lerdge XI