Ultimaker/Cura #7471: Settings Header Text does not fit

gizmocuz, 2020-04-11:
**Application version**
4.6 (Beta)

**Platform**
Windows

**Printer**
Ender-3

**Reproduction steps**
When setting Infill to 100% this always happens

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 
[image]

**Expected results**
All labels to fit

**Additional information**
Desktop resolution 4K, Scale/Layout 125%

Ghostkeeper, 2020-04-21:
Hmm, I'm unable to reproduce this, but my operating system only supports 100% or 200%. Can't wait for Ubuntu 20.04 to hit the day after tomorrow...

gizmocuz, 2020-04-21:
@Ghostkeeper , it might be possible to install virtualbox and emulate another operating system ;)

Ghostkeeper, 2020-04-21:
Last weekend I built an entire [C++ application](https://github.com/Ghostkeeper/ConvertTo3mf) from scratch because I didn't feel like rebooting to Windows. I'm just lazy, sorry. I'm actually hoping that someone that develops on Windows can test this, secretly.

gizmocuz, 2026-05-31:
No it is perfect now. Thank you for all your work!