Ultimaker/Cura #7474: Preview-simulator play button not appearing in 4.5.0 Stable & 4.6.0 beta

Welsh-Dragon, 2020-04-11:
O/S Win 7 Home Premium 64 bit Acer Laptop, 3 GB RAM, ATI Radeon HD 3200 Graphics, CPU AMD K8

Steps taken;

~0 Open Cura from Start Menu, [it takes about 1 minute to load]
~1 Click on Get Started
~2 Click on Agree
~3 Click on Next
~4 Choose Add a non networked printer
~5 Scroll down & choose Prusa i3 Mk3/Mk3s
~6 Click next
~7 Click Finish
~8 Open file(s)
~9 Locate & open selected .stl file, [does not matter which one]
~10 Click on Slice
~11 Click on Preview via the Preview-Save File box, in the lower right corner

As you can see the Slider is on the right side of the screen, & it works as should, however, the Play Simulator button is not showing up.

[image]

I am getting the issue in both 4.5.0 stable & 4.6.0 Beta

I have also chosen the default Custom printer in 4.5.0 Stable & the Play-Simulator button is still MIA...

I am very grateful for you looking into this issue...

Kind regards & stay-safe & well...

**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system and maybe GPU.)

**Printer**
(Which printer was selected in Cura?)

**Reproduction steps**
1. (Something you did.)
2. (Something you did next.)

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/Ultimaker/Cura#logging-issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

smartavionics, 2020-04-11:
Your graphics are in compatibility mode because either you have explicitly set that in the cura preferences or, more likely, your graphics system (HW+drivers) does not provide OpenGL 4.1 or higher. Try updating your graphics drivers.

fieldOfView, 2020-04-11:
In addition to what @smartavionics said, your GPU supports up to OpenGL 3.3, so no matter what driver you install it does not meet the requirements for the “enhanced” simulation view.

Welsh-Dragon, 2020-04-11:
If it were a PC I could likely get an upgrade to the GPU but a laptop, I doubt it tbh...

Thank you both @fieldOfView & @smartavionics  for your prompt replies, I really appreciate it...

Is there a print-simulator online, that could be used as a workaround, till I can afford to get a new compatible machine...??

smartavionics, 2020-04-11:
You could try installing the crafware slicer because it has a very nice gcode viewer that is far better than the cura simulator anyway.

smartavionics, 2020-04-11:
sorry, that should be craftware

Welsh-Dragon, 2020-04-11:
Thank you @smartavionics , I will give it a go & hope the GPU is compatible with it...

Welsh-Dragon, 2020-04-11:
I have come across https://craftbot.com/craftware/ & it offers two programs

CraftWare & CraftPrint

I am not sure at this stage which one I need or is it both...??

smartavionics, 2020-04-11:
I think craftbot is their printer? You just need craftware

Welsh-Dragon, 2020-04-11:
Its a bit offputting as they need you to have an account prior to download, which negates the **free-download** part...

Ghostkeeper, 2020-04-21:
This is not something we're likely to fix. The old "compatibility" layer view simply draws the polygons it gets from the slicing process. The new one uses geometry shaders to draw the 3D tubes, and also to cut off the layer at a certain point. The geometry shaders are not available for your GPU.

closed 2020-04-21 by Ghostkeeper