Ultimaker/Cura #7475: gcode visualisation

Michael26K, 2020-04-11:
Cura 4.5.0, Preview, Layer view.

After hitting the "Play" button I can see the nozzle moving. But I can not see the exact position.
I want to see the exact start / end position of the movement to compare the position with the printing result (dimensions). Actually I am using the saved gcode and look in a text editor to find the positions which costs time and is not very comfortable.

Nice would be to have a window on the screen which shows the gcode in a scrolling mode.
- highlighted: the actual line of the running simulation on a fixed position in the center of the window
- previous and upcomming 5 to 10 lines of gcode above / below.
(pic attached)

Alternatively / additionally it could be the exact position of the nozzle (x / y / z coordinates) as the z coordinate is not visible in the gcode at this time.

The gcode / position should still be visible if the simulation is stopped / paused.

It would help me to find the exact position / dimensions of the noozle and comepare the real dimensions of the printed part and the gcode to narrow discrepancies without saving, transferring and searching the gcode in an extra program.

It might help people also to understand and follow gcode and printing as a teaching feature.

(https://user-images.githubusercontent.com/63508123/79048181-8aa12800-7c1b-11ea-9ec4-4866c6a843b7.PNG)

smartavionics, 2020-04-11:
The craftware slicer has a gcode viewer mode that does most of what you are asking.

Ghostkeeper, 2020-04-21:
I tend to use Repetier-Host for this purpose, as another alternative.

mahtDFR, 2020-04-22:
Thanks for the suggestion. After some discussion this morning we have decided we don't have the capacity to do this right now, so it is deferred.

closed 2023-07-13 by github-actions[bot]