Ultimaker/Cura #7476: 4.5 and 4.6 beta Very slow response from GUI

sonnenbs, 2020-04-11:
I have not been able to do much with Cura.  I have been able to load models from others and slice them after being patient but I would like to do my own models.  The GUI is very unresponsive which makes it horrible to use.

**Application version**
I experienced this in both version 4.5 and 4.6 beta.

**Platform**
Windows 10 pro and tried Windows 8 compatibility mode.

**Printer**
Ender-3

**Reproduction steps**
After setting anything in the profile, trying to rotate the model, clicking between views, etc.  Pretty much any time I tried to do something with the GUI.

**Screenshot(s)**
N/A

**Actual results**
Takes 10 - 60 seconds to respond.  

**Expected results**
It should respond much quicker.

**Project file**
N/A

**Log file**
2020-04-11 01:41:57,418 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-11 01:42:16,825 - INFO - [Thread-3] STLReader.STLReader.load_file [52]: Using NumPy-STL to load STL data.
2020-04-11 01:42:16,837 - DEBUG - [Thread-3] UM.Mesh.MeshData.calculateNormalsFromVertices [548]: Calculating normals took 0.0019440650939941406 seconds
2020-04-11 01:42:16,838 - DEBUG - [Thread-3] STLReader.STLReader._read [104]: Loaded a mesh with 73386 vertices
2020-04-11 01:42:16,861 - DEBUG - [Thread-3] UM.Mesh.MeshData.approximateConvexHull [503]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.018965482711791992 seconds. 80 input vertices. 80 output vertices.
2020-04-11 01:42:16,874 - DEBUG - [Thread-3] UM.FileHandler.ReadFileJob.run [71]: Loading file took 0.0 seconds
2020-04-11 01:42:16,897 - DEBUG - [Thread-3] UM.Mesh.MeshData.approximateConvexHull [503]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.019946813583374023 seconds. 80 input vertices. 80 output vertices.
2020-04-11 01:42:16,930 - DEBUG - [MainThread] UM.Operations.OperationStack.push [72]: <UM.Operations.AddSceneNodeOperation.AddSceneNodeOperation object at 0x0000026E5F541080>, took 1ms
2020-04-11 01:42:17,161 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [c:\program files\ultimaker cura 4.6\resources\shaders\transparent_object.shader]...
2020-04-11 01:42:25,294 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [58]: Loading shader file [c:\program files\ultimaker cura 4.6\resources\shaders\toolhandle.shader]...
2020-04-11 01:42:26,940 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-11 01:43:13,278 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-04-11 01:43:13,281 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-11 01:43:14,013 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [139]: Need to check for errors again. Discard the current progress and reschedule a check.
2020-04-11 01:43:14,175 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [135]: New error check scheduled.
2020-04-11 01:43:15,336 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [187]: Error check finished, result = False, time = 1.2s
2020-04-11 01:43:23,501 - DEBUG - [MainThread] cura.AutoSave._onTimeout [60]: Autosaving preferences, instances and profiles
2020-04-11 02:21:17,645 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.triggerNextCallback [51]: No more on-app-exit callbacks to process. Tell the app to exit.
2020-04-11 02:21:17,662 - INFO - [MainThread] cura.CuraApplication.closeApplication [577]: Close application
2020-04-11 02:21:17,674 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [470]: Shutting down cura
2020-04-11 02:21:17,708 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [304]: Attempting to kill the engine process
2020-04-11 02:21:17,709 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [310]: Killing engine process
2020-04-11 02:21:17,718 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [313]: Engine process is killed. Received return code 1

**Additional information**
I have tried searching for fixes.  I tried all of the things they that were suggested like turning off many of the plugins, running in compatibility mode, changing firewall settings, and changing preferences.

rburema, 2020-04-14:
Was this introduced in 4.5, or do 4.4 and earlier also give problems? If not, does 4.4 or earlier give problems when you start them up _now_?

sonnenbs, 2020-04-14:
Hi rbuerma,

Thank you for responding.  I have not tried it on an earlier version of Cura.  I just got the printer and was my first time attempting to use Cura.  I was not sure where I could get an earlier version of the software.

rburema, 2020-04-16:
Ah I just assumed it was in comparison to earlier versions.

It might be the USB plugin. If you don't print via USB, you can disable that plugin from the marketplace (button next to sign-in in the top-right. Otherwise, what's your graphics card?

sonnenbs, 2020-04-17:
Hi rburema,

I have already disabled the USB plugin.  It looks like I have a NVIDIA Quadro P1000.

Ghostkeeper, 2020-04-21:
I'm unable to reproduce this. Generally everything responds within 30ms or so. A couple of notable exceptions to this are:
* Changing the scene if there are many models in the scene. This is still a known bottleneck.
* Loading big files.
* Switching printers.
* Changing something in the printer settings that warrants a recalculation of the lists of profiles.
* Changing materials in such a way that warrants a recalculation of the lists of profiles.
* Starting a slice with many models on the build plate with adjusted extruders or per-object settings.

The 1st and last of these are something we'll probably be taking a look at for the next release.

I don't think this is the performance of your graphics card per sé unless the frame rate is also bad while rotating the camera around in layer view.

sonnenbs, 2020-04-22:
Hi Ghostkeeper,

Unfortunately none of your bullets apply in my case.  

Thanks for your feedback.

closed 2020-06-19 by no-response[bot]

IMBalENce, 2020-09-16:
I have also been experiencing this issue with all latest versions from 4.5 to the current 4.7.1. GUI has been super sluggish and frozen every 10-20 seconds. Frustrated for a few weeks until I found this post 
https://community.ultimaker.com/topic/29372-cura-running-super-slow-on-a-really-good-computer/page/2/

**How to Fix Issue:**
- Open Cura and enter the Marketplace at the top right in application.
- Navigate to the "Installed" tab when in the market place. 
- Scroll to find the plug in titled "UM3 Network Printing" and just uncheck the box.
- I don't suggest deleting plugin just in case there is a reason for you to utilize it in the future.
- You will now see a new blue button in the marketplace at the bottom right - "Quit Cura"
- Quit Cura and restart. If this was your problem, Cura should now be running smoothly. 

Instantaneous fix for me.

lasmaty07, 2020-10-04:
> I have also been experiencing this issue with all latest versions from 4.5 to the current 4.7.1. GUI has been super sluggish and frozen every 10-20 seconds. Frustrated for a few weeks until I found [...]

I'm running Cura 4.7.1 and this definitely fixed the issue for me. 
Another tip: you should disconnect from you LAN in order to do those steps quicker, as soon as you turn of you wifi or disable your ethernet connection, cura start running smoothly.