Ultimaker/Cura #7478: [4.5] Spiralize Outer Contour ignores Speed Settings

Keichide, 2020-04-12:
**Application version**
4.5

**Platform**
Windows 10 

**Printer**
Creality Ender 3

**Reproduction steps**
1. Get a Vase from Thingiverse
2. Set Wall Speed to 45mm/s
2. Slice it with Spiralize Outer Contour enabled

**Actual results**
Cura is ignoring the Speed Settings and useses the "Minimum Speed" Setting instead for the whole Print

**Expected results**
Use the entered Speed Values in the Speed Tab for the Spiralize Mode instead or notify or at least put a Information somewhere which Values it uses.

smartavionics, 2020-04-12:
Reduce the min layer time so that the speed reduction doesn't occur.

Keichide, 2020-04-12:
Yeah.. That was the Issue.. But anyway... Please at least right it down somewhere what affects the Speed of the Spiralize Mode.

closed 2020-04-12 by Keichide

Ghostkeeper, 2020-04-21:
It says it affects the speed of everything, as documented.