Ultimaker/Cura #7479: 4.5 Fill Gaps Between Walls Fills Buffer

BritNoob, 2020-04-12:
Cura 4.5
Hypercube Evolution - MKS Gen 1.4, RPi, Octoprint, Klipper

Having an issue where the printer fills its buffer with "Fill Gaps Between Walls" causing the printer to stop and make a blob while the head is stopped. It also shakes the printer quite violently i'm guessing because of the many tiny movements over a very short period of time it fills the buffer causing the crash.
I have tested this issue and with different configs and have found that reducing the speed does help but then it affects the rest of the print time.

Is there a way of adding an independent speed option associated with "Fill Gaps Between Walls" rather than making it a factor of the normal print speed. 
Also rather than doing a Zig Zag down the gap most of these gaps could normally be filled by a single line down the middle. Is there some way of adding this as an option?

Thanks!

smartavionics, 2020-04-12:
I understand that the UM devs are working on a new implementation of wall printing that should fix your issue. In the meantime, you may wish to install one of my cura builds that feature a better implementation of gap filling then the current UM releases. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there before using.

Ghostkeeper, 2020-04-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/4871. Indeed we're making some attempts to fix this with a new algorithm.

closed 2020-04-21 by Ghostkeeper