Ultimaker/Cura #7480: Cura 4.5 Manual setting of Z location is overridden

Wdfitzpat007, 2020-04-12:
**Application version**
4.5

**Reproduction steps**
1. Tried to lift a model off the build plate to have it print over support material
2. Watched it snap back to the build plate
3. Manually set Z height
4. Watched the model snap back down.
5. Changed the model to "change settings overlap"
6. Enabled lock model
7. Manually entered Z height
8. Watched the model snap back to the build plate
9. Cured corona virus

**Actual results**
Z height reset to 0

**Expected results**
Z height persist

Note that I will be opening reports for the group and merge features for similar, but separate, issues.

smartavionics, 2020-04-12:
You have, of course, unchecked the drop models on buildplate option in the Cura prefs?

[image]

Wdfitzpat007, 2020-04-13:
Ill re-open this as a feature request, IMHO drop models should not be hidden in cura prefs. If you have a manual set Z option in the move object menu, having it disabled but appearing active to the user is very frustrating.

Wdfitzpat007, 2020-04-13:
On second thought, if you have a user input that appears active, but is in-fact disabled, that is a bug.

Ghostkeeper, 2020-04-21:
We've enabled the drop to build plate by default because typically if you lift the model off the build plate it would ruin your print. So to prevent new users from accidentally ruining their print it should be enabled. You can however still lower the print into the build plate, which is why the widget is still activated in the translate tool.

Sounds to me like this is behaving as expected.

closed 2020-04-21 by Ghostkeeper