Ultimaker/Cura #7481: Cura 4.5 Grouping and Merging locked models resets model location

Wdfitzpat007, 2020-04-12:
**Application version**
4.5

**Reproduction steps**
1. Attempted to group/merge two support blocker models
2. Watched them snap to Z 0 and to the center of the build plate.
3. Attempted to undo, but the original model location was permanently lost
4. Moved models where they belong
5. Re-attempted this same procedure with the models locked
6. Models changed location again, this time staying at the same xy location but appearing below the build plate.
7. Attempted to undo, but the original model location was permanently lost
8. Pillage a village out of spite

**Expected results**
Models should not move... duh. Also it would be nice if undo worked :P

Would write more detailed bug reports, but these are painfully basic functionality bugs.

Hey, if the merge tool worked for support blockers, or if you could edit the Z location of models. It would make the cutting mesh of disappointment actually useful. Would become a super powerful tool.

rburema, 2020-04-12:
Retry with the 'drop to buildplate' preference off?

Wdfitzpat007, 2020-04-13:
Yes, Ill re-post this as a feature request to have the drop to build-plate setting to be added to the move object section.

Wdfitzpat007, 2020-04-13:
Oops, no. Merged models simply pop to the centerpoint of the buildplate. XY and Z cords are reset. Undo does not work.

nallath, 2020-04-17:
If you merge two models, they always snap. That's the point of the merging after all. Group ensures that the models remain on the same position on the build plate, whereas merge ensures that the origins of the two models are set to be the same (Merge is to ensure that dual extrusion models can be printed correctly)

So if you group the models, you can correctly undo it (and they remain at the same spot).

closed 2020-04-17 by nallath