Ultimaker/Cura #7482: [master] signing in cause cura to use 100% CPU

smartavionics, 2020-04-12:
**Application version**
master branches as of 12th April 2020

**Platform**
Linux

**Printer**

Creality

**Reproduction steps**
Sign in to account.

**Screenshot(s)**

**Actual results**
Sign in appears to happen but then Cura continues to run at 100% CPU (1 core busy) forever.

**Expected results**
After doing the sign in, Cura should slumber as per normal.

**Project file**

**Log file**

**Additional information**
Couldn't see any obvious error/warning messages that related to sign in from either stderr or cura.log.

Ghostkeeper, 2020-04-21:
I'm not able to reproduce this so far.
[image]

I guess we'll have to keep an eye out for it. If we know how to reproduce the issue we can run a profiler on it to see what it's busy doing.

smartavionics, 2020-04-21:
Here's some profile data...

[yappiProfile.zip](https://github.com/Ultimaker/Cura/files/4511541/yappiProfile.zip)

[image]

Ghostkeeper, 2020-04-22:
Hmm, the `_handle_request_noblock` is not a function of ours. We're not calling it directly. Neither `get_request` or `process_request`. Seems that something froze in the webserver that's waiting for a response from the login form, maybe?

nallath, 2020-06-30:
We quite recently fixed this (so after signing in, it didn't close the signing in server correctly)

closed 2020-06-30 by nallath